realsenseai/librealsense #11844: keeps showing no frames received.

Pradeep-Sudakar, 2023-05-26:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|233622075978|
|**Firmware Version**|05.14.00.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|226523131892|
|**Firmware Update Id**|226523131892|

Please provide a description of the problem

MartyG-RealSense, 2023-05-26:
Hi @Pradeep-Sudakar  Could you provide more information about your problem, please?  For example, is the No Frames Received message occurring in the RealSense Viewer tool?  Thanks!

Pradeep-Sudakar, 2023-05-26:
The camera works on another laptop but its not working on mine

Pradeep-Sudakar, 2023-05-26:
And yes the No frames message is occurring on the RealSense Viewer tool

Pradeep-Sudakar, 2023-05-26:
[image]

Pradeep-Sudakar, 2023-05-26:
Could it be because of my anti-virus?

MartyG-RealSense, 2023-05-26:
It is possible.  I recall a past case of a RealSense user on Windows whose camera was being negatively affected by their antivirus software.  Microsoft also advise that antivrus software can block a camera.

https://support.microsoft.com/en-us/windows/camera-doesn-t-work-in-windows-32adb016-b29c-a928-0073-53d31da0dad5

Pradeep-Sudakar, 2023-05-26:
I went and checked my firewall settings as well nothing seems to be blocked

[image]

Pradeep-Sudakar, 2023-05-26:
I went and checked my firewall settings as well nothing seems to be blocked

[image: Screenshot 2023-05-26 .png]

-- 
**Disclaimer:*
This message was sent from Vellore Institute of Technology.  
The contents of this email may contain legally protected confidential or 
privileged information of “Vellore Institute of Technology”.  If you are 
not the intended recipient, you should not disseminate, distribute or copy 
this e-mail. Please notify the sender immediately and destroy all copies of 
this message and any attachments. If you have received this email in error, 
please promptly notify the sender by reply email and delete the original 
email and any backup copies without reading them.*

MartyG-RealSense, 2023-05-26:
Is the camera able to work with the **rs-capture** example program that will have been installed when you installed the RealSense SDK on Windows?  Instructions for running rs-capture can be found at https://github.com/IntelRealSense/librealsense/issues/7544#issuecomment-707577574

Pradeep-Sudakar, 2023-05-26:
No I've tried that it didnt work. I just get an empty black screen and the program exits after a while

MartyG-RealSense, 2023-05-26:
Thank you.  Does **rs-hello-realsense** run?  This is a text-only example, so if it correctly updates the depth then it could indicate that there is a graphics related issue on that particular laptop, such as the graphics drivers needing to be updated.

Pradeep-Sudakar, 2023-05-26:
No it does the same thing just shows an empty black screen and terminates

MartyG-RealSense, 2023-05-26:
A black screen in these examples suggests that the camera is unable to be accessed by the programs.

Please check the Windows privacy settings using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789 to confirm that permission to access the camera and permission for the camera to access applications are both set to On.

Pradeep-Sudakar, 2023-05-26:
I have had it on but it still doesnt seem to work.

[image]

What else should i try?

Pradeep-Sudakar, 2023-05-26:
[image]

The camera is being accessed by the apps

MartyG-RealSense, 2023-05-26:
Please try unplugging the micro-sized end of the USB cable from the side of the camera, reversing the connector's orientation and plugging it back into the side of the camera (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

Pradeep-Sudakar, 2023-05-26:
No it is not still working. I have also tried deleting and reinstalling the drivers and updating the drivers but that also didnt seem to work

MartyG-RealSense, 2023-05-26:
Something that can cause these effects in the Viewer and the example programs is if a RealSense program is already open and using the camera when another program is launched.  For example, if rs-capture had been launched first and then the Viewer was opened secondly, when the depth stream is enabled then the No Frames Received message is displayed.

Can you confirm that no other RealSense program is running?

Pradeep-Sudakar, 2023-05-26:
Yes no other programs are running in the background.

Now its also saying this new error:
notifications.cpp:511 - Motion Module force pause

[image]

MartyG-RealSense, 2023-05-26:
There may be something in your laptop's hardware that the camera is conflicting with.  RealSense 400 Series cameras can work with any Intel or Arm processor chip, but success with AMD is rarer.  If your laptop has an AMD processor then that could cause problems.

Pradeep-Sudakar, 2023-05-26:
I have i7-12650H processor. That shouldnt be a problem right?

MartyG-RealSense, 2023-05-26:
i7 Intel Core should be fine.

Is the other laptop that the camera works on the same as your laptop?

Pradeep-Sudakar, 2023-05-26:
No the other one has a AMD Ryzen 5 processor.

MartyG-RealSense, 2023-05-26:
A laptop that has an i7 12th generation processor would likely have all USB 3 ports, so it is noticable that the camera is being detected as being on a slower USB 2.1 connection.

The camera, USB cable and camera firmware driver should be fine as the camera works on the other laptop.

I note that you have already tried uninstalling and reinstalling the RealSense drivers.

If the example programs do not work then downloading and running the standalone Viewer **Intel.RealSense.Viewer.exe** is not likely to work.

When Windows 10 downloads and installs updates it is possible for one of those updates to disrupt a RealSense camera.  It would not be possible to know which particular update had caused the problem though.

Pradeep-Sudakar, 2023-05-27:
Can having windows 11 be the problem? Should I downgrade and try it again?

-- 
**Disclaimer:*
This message was sent from Vellore Institute of Technology.  
The contents of this email may contain legally protected confidential or 
privileged information of “Vellore Institute of Technology”.  If you are 
not the intended recipient, you should not disseminate, distribute or copy 
this e-mail. Please notify the sender immediately and destroy all copies of 
this message and any attachments. If you have received this email in error, 
please promptly notify the sender by reply email and delete the original 
email and any backup copies without reading them.*

MartyG-RealSense, 2023-05-27:
RealSense usually works fine with Windows 11 though there were a small number of cases soon after the release of Windows 11 that had problems, such as the No Frames Received one at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/15149975945363-realsense-SDK-on-windows-11

Pradeep-Sudakar, 2023-05-31:
I tried everything that the post said it still doesnt seem to work. What should I try next?

MartyG-RealSense, 2023-05-31:
Let's first summarize what has been tried and found.

Using Windows 11 with camera firmware driver version 5.14.0.0..

No Frames received in the Viewer.  Blank screen in example programs such as rs-capture and rs-hello-realsense.

Firewall is not blocking.

Reversing the orientation of the micro-sized connector on the USB cable did not help.

Deleting the drivers and reinstalling them did not help.

The camera is detected as USB 2.1 and not USB 3.

There is not another RealSense program open already.

Windows camera privacy options are both set to On.

The laptop on which the problem occurs has an Intel i7 processor and the laptop that it works on is AMD Ryzen.  So the processor type is likely not the problem.

*******************

So let's focus next on the USB connection type.  On the laptop with the problem, the USB connection is detected as USB 2.1.  On the laptop that the camera works correctly on, is it a USB 3 connection?

Pradeep-Sudakar, 2023-06-01:
Yes all the ports are 3.0 

[image]

Pradeep-Sudakar, 2023-06-01:
Should i maybe try using a different USB cable because the one I'm using is not the one that was provided by intel?

MartyG-RealSense, 2023-06-01:
I would recommend trying that if you are not using the official 1 meter cable.  It is important to use a high quality USB cable with RealSense cameras because of the volume of data that they transmit.  It should be a cable designed for data transfer and not an inexpensive charging / sync cable.

MartyG-RealSense, 2023-06-06:
hi @Pradeep-Sudakar  Do you require further assistance with this case, please?  Thanks!

Pradeep-Sudakar, 2023-06-08:
Not as of now @MartyG-RealSense thank you so much for all your help

MartyG-RealSense, 2023-06-08:
You are very welcome, @Pradeep-Sudakar - thanks very much for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-06-08 by MartyG-RealSense