realsenseai/librealsense #11846: T261 camera shows "No Frames Received"

TexenJin, 2023-05-28:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|
Here is the erro info
01:59:38 [Info]  - librealsense version: 2.53.1

01:59:38 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2701 - Framebuffer size changed to 1895 x 1015
01:59:38 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2701 - Window size changed to 1895 x 1015
01:59:38 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2701 - Scale Factor is now 1
01:59:38 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1285 - Failed to add Backlight Compensation control for RGB Camera
01:59:38 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1285 - Failed to add Exposure control for RGB Camera
01:59:39 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1285 - Failed to add Enable Auto Exposure control for RGB Camera
01:59:39 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1285 - Failed to add Backlight Compensation control for RGB Camera
01:59:39 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1285 - Failed to add Hue control for RGB Camera

Please provide a description of the problem

TexenJin, 2023-05-28:
[image]
[image]

dmipx, 2023-05-29:
Hi @TexenJin 
Unfortunately, "platform camera" is not supported by RealSense viewer as it's not RealSense camera, but integrated webcam.
This cannot work.

TexenJin, 2023-05-29:
The camera was worked,all because the wrong connecting band direction (;¬_¬)
I going to close this issue,thanks for your answer

closed 2023-05-29 by TexenJin