realsenseai/librealsense #11847: OpenVINO and RealSense SDK

A-Robotics-mama, 2023-05-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455                                                      | 
| Firmware Version                   |                                                                 | 
| Operating System & Version |   Win 11                                                   |
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                 | PC                                                           |             
| SDK Version                           |  SDK WIN10-2.53.1.4623                        | 
| Language                               |  python                                                  | 
| Segment	         	      	|  Robot                                                     | 

### Issue Description
Please advise about compatibility SDK with last version of OpenVINO 2022.3

MartyG-RealSense, 2023-05-29:
Hi @A-Robotics-mama  Support for OpenVINO 2020 and 2021 was added in RealSense SDK version **2.49.0** and newer.  There is not currently official support for the 2022 version.

A-Robotics-mama, 2023-05-31:
So, no sense to install last OpenVINO for work with SDK  2.53?

MartyG-RealSense, 2023-05-31:
The RealSense OpenVINO wrapper's GitHub page states that "RealSense examples have been designed and tested with the following OpenVINO versions: 2019.3.379, 2020.3.194, 2020.4.287, 2021.1.110, 2021.3.394.  Working with other versions may require code changes".

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino#openvino-samples-for-intel-realsense-cameras

I believe that it is referring to changing the code of the examples, not the code of the wrapper.

So OpenVINO has been tested up until version 2021.3.394.  You are the first RealSense user who has enquired about using OpenVINO 2022.  So the examples may be able to work with 2022.3 if you try it but there may be errors, in which case it could be tried with 2021.3.394 if those errors could not be resolved thorugh code changes.

closed 2023-06-12 by MartyG-RealSense