realsenseai/librealsense #11848: No module named 'librealsense'

huyuaaaray, 2023-05-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {MacOS  | 
| Kernel Version (Linux Only)    |  (N/A)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.53.1 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
<when i import the pyrealsense2 and trying to use the api to capture the frames of the camera in Python. It works well on win, but when i use the script in mac, it will return:

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Cell In[6], line 12

---> 12 import librealsense as rs
     13 import numpy as np
     14 from ultralytics import YOLO

ModuleNotFoundError: No module named 'librealsense'>

it seems it cannot find the camera, but i check the mac device info, the camera exists.

MartyG-RealSense, 2023-05-29:
Hi @huyuaaaray  There is a pre-built version of pyrealsense2 for Mac at https://github.com/IntelRealSense/librealsense/issues/9687 but it has only been tested with MacOS Catalina and Big Sur.

At the time of writing this, the most recent MacOS version that has pyrealsense2 setup advice available for it is **Monterey** - see https://github.com/IntelRealSense/librealsense/issues/5275#issuecomment-1059876482

A RealSense user who recently tested pyrealsense2 with the current latest MacOS, **Ventura**, found that the camera was not detected.

MartyG-RealSense, 2023-06-06:
Hi @huyuaaaray  Do you require further assistance with this case, please?  Thanks!

huyuaaaray, 2023-06-06:
thanks， so is there any way to use the camera in Mac?

获取 Outlook for iOS<https://aka.ms/o0ukef>

[External Email]

MartyG-RealSense, 2023-06-06:
The camera and the librealsense SDK can be used with MacOS Monterey and Apple Silicon M1 based Macs.  It can be installed with **brew install librealsense** or by using a detailed installation guide at the link below.

https://lightbuzz.com/realsense-macos/

The guide includes a link to an automated shell script for installing pyrealsense2 on MacOs.

https://github.com/cansik/pyrealsense2-macosx/blob/apple-silicon/librealsense-python-mac.ps1

At the time of writing this there is not much information about using RealSense with MacOS Ventura other than a recent case at https://github.com/IntelRealSense/librealsense/issues/11815

huyuaaaray, 2023-06-07:
Hi Marty,
I do can install the pyrealsense on Mac. Like the description previously, how can I solve the problem that the system cannot detect the camera?
Thanks,
Yu Hu

获取 Outlook for iOS<https://aka.ms/o0ukef>

[External Email]

https://lightbuzz.com/realsense-macos/

https://github.com/cansik/pyrealsense2-macosx/blob/apple-silicon/librealsense-python-mac.ps1

MartyG-RealSense, 2023-06-07:
I note that the instruction ** import librealsense as rs** is mentioned, producing the error **ModuleNotFoundError: No module named 'librealsense'**.

In a pyrealsense2 script, the instruction should be **import pyrealsense2 as rs**

huyuaaaray, 2023-06-07:
Thanks, I do import the pyrealsense2, but it still cannot detect the device.
[image]
[image]
[image]

MartyG-RealSense, 2023-06-07:
Is the **realsense_config** in the pipeline.start() brackets a cfg stream configuration name?

If it is then a custom configuration would not be written like this with the width, height and framerate inside the brackets.  It would usually look something like this when written in Python language:

```
pipeline = rs.pipeline()
realsense_config = rs.config()
realsense_config.enable_stream(rs.stream.depth, width. height, rs.format.z16, framerate)
realsense_config.enable_stream(rs.stream.color, width. height, rs.format.rgb8, framerate)
pipeline.start(realsense_config)
```

huyuaaaray, 2023-06-09:
[image]
Thanks, here is my config. I notice the code you provide has a format of `width. height`, but it throws an error when i edit to that. When I use `width, height`, it still shows no device detected:
[image]

MartyG-RealSense, 2023-06-09:
You are correct, there was a typo error in my provided code.  It should have been a comma beside width instead of a full stop.  My apologies, I have edited the comment above.

```
realsense_config.enable_stream(rs.stream.depth, width, height, rs.format.z16, framerate)
realsense_config.enable_stream(rs.stream.color, width, height, rs.format.rgb8, framerate)
```

I note that You still had **No device detected** error even after making the edit though, and that the problem only occurs on your Mac installation of pyrealsense2 but not the Windows one.

Have you tested other RealSense programs, such as realsense-viewer, to see whether they can detect the camera on Mac?

huyuaaaray, 2023-06-09:
I have tried the realsense-viewer, it will automatically quit if i connect to the camera. But it won't quit the exe if i don't connect the camera.

MartyG-RealSense, 2023-06-09:
Which MacOS version is your computer using, please?

Intel ceased updating RealSense Mac compatibility a long time ago after 2018's MacOS High Sierra version.  The RealSense SDK has continued working with subsequent MacOS versions since then.  But problems started to occur after the introduction of Apple Silicon M1 Macs and MacOS Monterey.  

As new versions of MacOs are released, such as the current latest **Ventura**, incompatibility may increase further, unfortunately.  The community of RealSense Mac users are now the best source for solutions to problems that arise with RealSense MacOS installation problems.  Many of those users can be found at https://github.com/IntelRealSense/librealsense/issues/9916

huyuaaaray, 2023-06-09:
It is Ventura, is there any way to solve this problem?
Is there any chance that intel will solve this problem, the compatibility of mac, later?

获取 Outlook for iOS<https://aka.ms/o0ukef>

[External Email]

MartyG-RealSense, 2023-06-09:
Intel no longer provide updates for Mac compatibility in the RealSense SDK.  Like with Monterey, a solution for Ventura will likely come from the community of RealSense users with Macs.  I recommend posting a comment on https://github.com/IntelRealSense/librealsense/issues/9916 to see whether somebody has a suggestion.

huyuaaaray, 2023-06-09:
I appreciate that

获取 Outlook for iOS<https://aka.ms/o0ukef>

[External Email]

closed 2023-06-09 by huyuaaaray