realsenseai/librealsense #11853: Debugging IMU/pipeline callback problem

sebasth, 2023-05-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |   2.53.1                          | 
| Language                            |  C++ (VS2022)                        | 
| Segment                             | |

### Issue Description

I have a problem with the pipeline callback when accelerometer stream is enabled. I am looking for any advice on how to debug this issue, as the sample code (`rs-callback.cpp`) works as expected when compiled as standalone executable. However the code doesn't work anymore once I use it in my application (on a specific laptop).

When accelerometer stream is enabled, the callback is not triggered anymore, unless `cfg.resolve(pipe)` is called after `start()` and then there are only a few motion frames (about 10 to 20) in the start, before any color/depth. Once the callback gets called with a color and depth frame, the callback doesn't receive further motion frames.

When accelerometer stream is not enabled, the callback gets called normally with the color/depth frameset (and it also works without calling `cfg.resolve(pipe)` )

I have also tested this on another PC, where the exactly same code works as expected (same camera, same SDK version, built with VS2019): both color/depth frames and motion frames are passed to the callback.

I would appriciate any suggestions on how I could debug/pinpoint the exact problem (why the code works on one PC but not on another).

sebasth, 2023-05-30:
The code I've used to test this
```
rs2::stream_profile select_profile(const std::vector<rs2::stream_profile>& profiles, int width, int fps,
	rs2_stream type = rs2_stream::RS2_STREAM_ANY) {
	
	for (auto& profile : profiles) {
		if ((type != rs2_stream::RS2_STREAM_ANY) && (profile.stream_type() != type)) { continue; }
		if (profile.is<rs2::video_stream_profile>()) {
			auto intrin = profile.as<rs2::video_stream_profile>().get_intrinsics();
			if (intrin.width <= width && profile.fps() <= fps) { return profile; }
		}
	}
}

// called only once
void run_realsense() {
	static rs2::pipeline pipe;
	static std::map<int, int> counters;

	rs2::context ctx;
	auto devices = ctx.query_devices();

	int  res_depth = 1024;
	int  res_colour = 1280;
	int  fps_depth = 30;
	int  fps_colour = 30;

	auto dev = devices.front();

	// configure device
	rs2::config cfg;
	auto sensors = dev.query_sensors();
	cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

	// find color and depth sensor and select parameters
	for (auto& sensor : sensors) {
		rs2::stream_profile profile;
		
		if (sensor.is<rs2::depth_sensor>()) {
			profile = select_profile(sensor.get_stream_profiles(), res_depth, fps_depth, rs2_stream::RS2_STREAM_DEPTH);
			cfg.enable_stream(RS2_STREAM_DEPTH, intrin.width, intrin.height, RS2_FORMAT_Z16, profile.fps());
		}
		else if (sensor.is<rs2::color_sensor>()) {
			profile = select_profile(sensor.get_stream_profiles(), res_colour, fps_colour);
			cfg.enable_stream(RS2_STREAM_COLOR, intrin.width, intrin.height, RS2_FORMAT_BGRA8, profile.fps());
		}
		else if (sensor.is<rs2::motion_sensor>()) {
			cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
			cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F);
		}
	}

	rs2::pipeline_profile profiles = pipe.start(cfg, [](const rs2::frame& frame)
		{
			if (rs2::frameset fs = frame.as<rs2::frameset>()) {
				for (const rs2::frame& f : fs) {
					LOG(INFO) << f.get_profile().stream_name() << ": " << counters[f.get_profile().unique_id()]++;
				}
			}
			else {
				LOG(INFO) << "IMU: " << counters[frame.get_profile().unique_id()]++;
			}
		}
	);
	
	auto profiles = cfg.resolve(pipe); // without this, no frames are passed to callback when accel stream is enabled	
}
```

MartyG-RealSense, 2023-05-30:
Hi @sebasth  Please first try adding **callback** to the brackets of the pipeline start instruction with no space between 'callback' and 'cfg' after the comma, like in the instruction below.  Putting callback in the bracket tells the program that it is a callback-using script.

`rs2::pipeline_profile profiles = pipe.start(callback,cfg, [](const rs2::frame& frame)`

sebasth, 2023-05-31:
I managed to pinpoint the issue to a 3rd party library portaudio. Everything seems to work after downgrading portaudio to older version 19.6.0 (from 19.7.0).

closed 2023-05-31 by sebasth

MartyG-RealSense, 2023-05-31:
It's excellent to hear that you achieved a solution.  Thanks very much for the update!