realsenseai/librealsense #11857: Memory leak on create and delete net_device

Zerpico, 2023-05-31:
SDK Version: v2.38.1.2223
OS: Windows 10 Home 64 bit
Platform: Laptop (Intel Core i7-10510U, 16G RAM, NVIDIA GeForce MX250 2G)
Language: C++ (Visual Studio 2019)
Segment: Desktop
Camera Model: D435i
Firmware Version: 05.12.07.100

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC (Intel Core i5-10500, 32GB RAM)  |
| SDK Version                           |  2.53.1                                                 | 
| Language                               |  C/C#                                                  | 
| Segment		          	|  others                                                 | 

### Issue Description

I'm using the camera on a raspberry pi with a rs-server, and a computer as a network client.
I noticed a problem with the memory of the project. When reconnecting to a device and creating a net_device, memory leaks occur. Apparently some objects are not cleared.
Code example:

```cpp
int main()
{
    static const int max_restarts = 200;
    static int restart_count = 0;
    rs2::context _ctx;

    for (; restart_count <= max_restarts; ++restart_count) 
    {
        std::cout << "START CONNECT #" << restart_count << std::endl;
        rs2::net_device *_dev = new rs2::net_device(ip_camera);
        _dev->add_to(_ctx);
        delete _dev;        
    }
    
    std::cin.get();
    return 0;
}
```

<details>
    <summary>Screen</summary>
    [image]
</details>

MartyG-RealSense, 2023-05-31:
Hi @Zerpico  Does the leak still occur if you shorten your rs2::net_device instruction to the one below?

` rs2::net_device _dev(ip_camera);`

Zerpico, 2023-06-01:
> Hi @Zerpico Does the leak still occur if you shorten your rs2::net_device instruction to the one below?
> 
> ` rs2::net_device _dev(ip_camera);`

I tried and the result is the same.
I also tried something like this
```std::shared_ptr<rs2::net_device> _dev = std::make_shared<rs2::net_device>(rs2::net_device(ip_camera));```

MartyG-RealSense, 2023-06-01:
If your aim in creating and deleting is to restart (as suggested by your max_restarts variable) then instead of using  `delete _dev; ` the documentation for the networking tool recommends using the command below once you have an SSH connection established with the IP address.

`sudo systemctl restart rs-server.service`

MartyG-RealSense, 2023-06-01:
There is not a precedent for resetting the network device by deleting it and recreating it, so it is difficult to conclude whether there is a bug in the networking tool.  

The tool is no longer updated though and is being removed from the SDK in its next version, so will not receive future fixes.  This is because it was an experimental proof of concept rather than an official, validated Intel RealSense feature.

MartyG-RealSense, 2023-06-07:
Version 2.54.1 of the RealSense SDK (librealsense) is now available at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

The 2.54.1 release notes state that the network device is removed in it so it would not be suitable to use for your networking project, though the notes add that there will be a replacement for the removed network device in the next SDK release.

MartyG-RealSense, 2023-06-13:
Hi @Zerpico  Bearing in mind the comments above, do you require further assistance with this case please?  Thanks!

Zerpico, 2023-06-13:
@MartyG-RealSense it's a pity that you decided to remove this functionality, I'll have to do it on my own. No, I don't need help anymore

MartyG-RealSense, 2023-06-13:
Intel are introducing a replacement for the removed network-device in the next SDK release according to the release notes of 2.54.1.  As 2.54.1 has only just been released at the time of writing this though, that next version may be some months away.

closed 2023-06-13 by MartyG-RealSense