realsenseai/librealsense #11866: Prebuilt binaries for legacy libraries

FrzMtrsprt, 2023-06-02:
| Required Info | |
|-|-|
| Camera Model | F200 | 
| Firmware Version | N/A | 
| Operating System & Version | Win 11 | 
| Kernel Version (Linux Only) | N/A | 
| Platform | PC |
| SDK Version | legacy | 
| Language | N/A | 
| Segment | N/A | 

### Issue Description
Where can I find the prebuilt binaries and the realsense viewer (if available) for the legacy models? I can't find them in the release or on the website.

MartyG-RealSense, 2023-06-02:
Hi @FrzMtrsprt  The legacy librealsense (librealsense1) that supports F200 can be found here:

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

There is not a RealSense Viewer for that SDK.

There was also an official SDK for Windows from 2016 that supported F200 but is not compatible with librealsense scripts.  A 1.8 gb installer program can be downloaded using this link:

http://registrationcenter-download.intel.com/akdlm/irc_nas/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe

FrzMtrsprt, 2023-06-02:
I will try this, thanks for the information!

closed 2023-06-02 by FrzMtrsprt

MartyG-RealSense, 2023-06-02:
You are very welcome!

If you try the 2016 Windows SDK then you should first run an installer program called **Depth Camera Manager (DCM)** before running the SDK installer program afterwards.  The DCM for the F200 can be downloaded from the Intel page below.

https://www.intel.com/content/www/us/en/download/18309/intel-realsense-depth-camera-manager.html