realsenseai/librealsense #11867: What is the return value of get_distance function ?

CheCamCheCam, 2023-06-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | v2.50.0| 
| Operating System & Version |   {Win 10  | 

| Platform                                 | PC  |
| SDK Version                            | ?                         | 
| Language                            |  C++                         | 
| Segment			|  others                        | 

### Issue Description
Hi everyone, I am making this issue because I have a doubt on what is the return value of the get_distance function from the rs2 library in C++, is it the Z coordinate value of a given point knowing the pixel associated, or is it the distance between the point and the origin of the base (which is the camera I believe) ?
Thank you for your time, best regards

MartyG-RealSense, 2023-06-02:
Hi @CheCamCheCam  The get_distance() instruction returns the real-world distance in meters between the camera and a coordinate in the scene.  

Its output is equivalent to taking the 2D depth pixel value (uint16_t) and multiplying it by the depth unit scale of the camera to obtain the distance from the camera to the coordinate in meters.  for example, if the raw depth pixel value was 6500 and the depth unit scale of the camera is 0.001 then the real-world distance will be 6.5 meters.

The default pixel depth scale of all RealSense 400 Series cameras except D405 is 0.001 (the D405's default depth scale is 0.01).

CheCamCheCam, 2023-06-05:
Hi @MartyG-RealSense thank you for your answer, so if I understand correctly if I want a representation of said coordinate in the 3D base of the camera I would still have to do more operations afterward, using the distance I just got right ?

MartyG-RealSense, 2023-06-05:
get_distance will give you the real-world Z-distance in meters.  If you needed all three parts of the world-space coordinate (X, Y and Z) then an efficient way to do that is to use the instruction **rs2_transform_color_pixel_to_depth_pixel**, which converts a 2D color pixel (XY) at a single specific coordinate to a 3D depth pixel (XYZ).

The base (underside) of the camera is not the 0,0,0 3D origin point of the coordinate systems.  The origin of depth is the center of the left IR sensor and the origin of RGB color is the center of the RGB sensor.

If depth to color alignment is performed then the 0,0,0 3D origin of the depth sensor changes to the center of the RGB sensor.

CheCamCheCam, 2023-06-06:
Hi @MartyG-RealSense I'm still confused about whether the get_distance will return the distance in 3 dimensions (implying that if we are talking about a coordinate not on the Z axis then the distance will increase according to Pythagora theorem) which is the one I am looking for, or the Z-distance (projection of 3D distance on the Z axis). Also in C++ I can"t seem to find information about the instruction "rs2_transform_color_pixel_to_depth_pixel", would you happen to have informations about it ?
Best regards

MartyG-RealSense, 2023-06-06:
get_distance returns the real-world Z distance in meters between the camera and a point in the real-world scene.  The coordinate whose value is provided is the one at the center of the image.

The depth pixel value is a measurement from the parallel plane of the imagers and not the absolute range, as illustrated in the image below.

[image]

The color to depth pixel conversion is **rs2_project_color_pixel_to_depth_pixel**.  My apologies, I have edited my answer in the comment above.  A C++ example of use of this instruction is at https://github.com/IntelRealSense/librealsense/issues/2982

CheCamCheCam, 2023-06-06:
Thanks a lot for the explanations, I do not have anything else to ask so i think I will close the issue here, have a good day

closed 2023-06-06 by CheCamCheCam