realsenseai/librealsense #11870: Unable to locate package librealsense2-utils E: Unable to locate package librealsense2-dev in Docker container Jetson

reinzler, 2023-06-04:
Running this Docker container results in

E: Unable to locate package librealsense2-utils
E: Unable to locate package librealsense2-dev

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455| 
| Firmware Version                   | | 
| Operating System & Version |   Ubuntu 20.04 , Docker container from this repo: https://github.com/dusty-nv/jetson-containers| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson Orin Nano  |

MartyG-RealSense, 2023-06-04:
Hi @reinzler  I note that you are using Ubuntu 20.04 on Jetson, which is an Arm-based board.  The librealsense SDK does not have packages for the Arm64 architecture for 20.04 and so librealsense has to be compiled for Arm from source code on 20.04 instead.

reinzler, 2023-06-04:
> Hi @reinzler I note that you are using Ubuntu 20.04 on Jetson, which is an Arm-based board. The librealsense SDK does not have packages for the Arm64 architecture for 20.04 and so librealsense has to be compiled for Arm from source code on 20.04 instead.

Hi, Marty, thanks for fast response! Can you provide a guideline, how to do it?

MartyG-RealSense, 2023-06-04:
A RealSense user provides a guide for building from source code with CMake on Nano and other Jetson board models at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

closed 2023-06-16 by MartyG-RealSense