realsenseai/librealsense #11872: I notice that if the object's too small / thin, it couldn't be detected, is it possible to configure the Intel Realsense D455 / program to detect thin and flat objects?

Nandomonaiya, 2023-06-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-06-05:
Hi @Nandomonaiya  If an object is thin then changing the **Depth Unit Scale** setting from its default value of 0.001 to **0.0001** may help to improve the image, as described at https://github.com/IntelRealSense/librealsense/issues/8228 when depth-sensing a toothpick and generating a pointcloud.

Thin objects can be difficult for the camera to see, as demonstrated in the recent case at the link below where high-up power lines were depth sensed with a D455 at a far range of around 20 meters with no background but open air behind them.

https://support.intelrealsense.com/hc/en-us/community/posts/17214523195155

In regard to minimum size, an object on the image should take up at least 9 pixels on the image in order to be trackable - please see https://github.com/IntelRealSense/librealsense/issues/4175

Nandomonaiya, 2023-06-06:
I will try the method, thank you. Also, is it possible to make the camera measures only length and width? therefore if the objects too small / thin, the height could be defaulted as a fixed variable

MartyG-RealSense, 2023-06-06:
If you are able to use Python then the **box_dimensioner_multicam** example may suit your needs.  It can automatically measure the length, width and height of an object and display the values in mm.  So you coud potentially check the length and width values and then default to height if the values are below a certain amount.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

The example requires that the camera is first pointed at a chessboard image placed on the floor and then place the object on the chessboard when prompted by the program so that the object can be measured.  

The purpose of the chessboard is to calibrate together the positions of multiple cameras arranged around the board, though if you are only using a single camera (the example works with just one camera) then it is possible to remove the need for the chessboard by editing the project's code to provide it with the intrinsics of your camera.  There is a discussion about this at https://github.com/IntelRealSense/librealsense/issues/9951

closed 2023-06-18 by MartyG-RealSense