realsenseai/librealsense #11875: Feature Request: On-device reset watchdog timer

michaeltandy, 2023-06-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415, and others I guess | 
| Operating System & Version |  All OSes, but I'm using Ubuntu 22.04  | 
| Platform                                 | All, but I'm using PC  |

### Issue Description

Please can we have [a watchdog timer](https://en.wikipedia.org/wiki/Watchdog_timer) that runs on the camera, controlled by the user's application, which restarts the camera if the timer times out?

There are plenty of ways you can get realsense cameras into funny, non-working states where you can't issue a reset command because you've lost communications. Power glitches, cable problems, rebooting while the camera's streaming, BIOS- and firmware- and kernel-dependent issues, all sorts of things.

A watchdog timer would provide a catch-all solution - and should reduce your support burden too.

MartyG-RealSense, 2023-06-06:
Hi @michaeltandy  I have highlighted your feature request to my Intel RealSense colleagues.  Thanks very much!

MartyG-RealSense, 2026-10-29:
Closing this case because the suggested feature request was not proceeded with by RealSense because of the large architectural changes to the SDK that would be required to implement it.

closed 2026-10-29 by MartyG-RealSense