realsenseai/librealsense #11877: ros2-fox; ubuntu20.04;sr305;d435i

TW050203, 2023-06-06:
**I tried "ros2 launch realsense2_camera rs_launch.py serial_no:=_635205000869" and "ros2 launch realsense2_camera rs_launch.py serial_no:=_635205000869 initial_reset:=true" for SR305 and it returns:**
[INFO] [launch]: All log files can be found below /home/rvc/.ros/log/2023-06-06-17-29-10-294039-rrvc-29961
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [29963]
[realsense2_camera_node-1] [INFO] [1686043750.570913990] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1686043750.571076302] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1686043750.571106846] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1686043750.595213763] [camera.camera]: Device with serial number 242322077348 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1686043750.595283536] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10.1/2-10.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1686043750.595307718] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1686043750.595938911] [camera.camera]: Device with port number 2-10.1 was found.
[realsense2_camera_node-1] [INFO] [1686043750.612095770] [camera.camera]: Device with serial number 635205000869 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1686043750.612164648] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10.2/2-10.2:1.2/video4linux/video8 was found.
[realsense2_camera_node-1] [INFO] [1686043750.612188748] [camera.camera]: Device with name Intel RealSense SR305 was found.
[realsense2_camera_node-1] [INFO] [1686043750.612752123] [camera.camera]: Device with port number 2-10.2 was found.
[realsense2_camera_node-1] [INFO] [1686043750.612786266] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1686043756.707174245] [camera.camera]: Device with serial number 242322077348 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1686043756.707281995] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10.1/2-10.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1686043756.707309207] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1686043756.707894012] [camera.camera]: Device with port number 2-10.1 was found.
[realsense2_camera_node-1] [INFO] [1686043756.724151352] [camera.camera]: Device with serial number 635205000869 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1686043756.724228022] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10.2/2-10.2:1.2/video4linux/video8 was found.
[realsense2_camera_node-1] [INFO] [1686043756.724248656] [camera.camera]: Device with name Intel RealSense SR305 was found.
[realsense2_camera_node-1] [INFO] [1686043756.724780411] [camera.camera]: Device with port number 2-10.2 was found.
[realsense2_camera_node-1] [INFO] [1686043756.727038659] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1686043756.727477955] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1686043756.727509407] [camera.camera]: Device Name: Intel RealSense SR305
[realsense2_camera_node-1] [INFO] [1686043756.727531315] [camera.camera]: Device Serial No: 635205000869
[realsense2_camera_node-1] [INFO] [1686043756.727550865] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10.2/2-10.2:1.2/video4linux/video8
[realsense2_camera_node-1] [INFO] [1686043756.727573118] [camera.camera]: Device FW version: 3.27.3.0
[realsense2_camera_node-1] [INFO] [1686043756.727591771] [camera.camera]: Device Product ID: 0x0B48
[realsense2_camera_node-1] [INFO] [1686043756.727608997] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1]  06/06 17:29:16,739 WARNING [140580092888832] (backend-v4l2.cpp:1444) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
[realsense2_camera_node-1]  06/06 17:29:16,739 WARNING [140580092888832] (backend-v4l2.cpp:1444) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
[realsense2_camera_node-1] [INFO] [1686043756.759948584] [camera.camera]: Set ROS param rgb_camera.profile to default: 1920x1080x30
[realsense2_camera_node-1] [WARN] [1686043756.772793778] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1686043756.773783778] [camera.camera]: Stopping Sensor: Coded-Light Depth Sensor
[realsense2_camera_node-1] [INFO] [1686043756.777941228] [camera.camera]: Starting Sensor: Coded-Light Depth Sensor
[realsense2_camera_node-1] [INFO] [1686043756.782595356] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1686043756.782717158] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1686043756.783882640] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1686043756.792895212] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1920, Height: 1080, FPS: 30
[realsense2_camera_node-1] [INFO] [1686043756.793486095] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1686043757.170589431] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1686043757.170709282] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.

**And I also tried "ros2 launch realsense2_camera rs_launch.py serial_no:=_242322077348" for D435i， it returns:**

[INFO] [launch]: All log files can be found below /home/rvc/.ros/log/2023-06-06-17-23-47-312550-rrvc-29826
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [29828]
[realsense2_camera_node-1] [INFO] [1686043427.659159109] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1686043427.659337603] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1686043427.659367274] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1686043427.683920087] [camera.camera]: Device with serial number 242322077348 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1686043427.684005178] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10.1/2-10.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1686043427.684033968] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1686043427.684619406] [camera.camera]: Device with port number 2-10.1 was found.
[realsense2_camera_node-1] [INFO] [1686043427.684651053] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1686043427.686940709] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1686043427.687434738] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1686043427.687477088] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1686043427.687502077] [camera.camera]: Device Serial No: 242322077348
[realsense2_camera_node-1] [INFO] [1686043427.687523645] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10.1/2-10.1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1686043427.687546195] [camera.camera]: Device FW version: 05.14.00.00
[... 5 lines cut ...]
[realsense2_camera_node-1] [ERROR] [1686043427.699504097] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)**
[realsense2_camera_node-1] [INFO] [1686043427.706681974] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[... cut ...]
[realsense2_camera_node-1] [INFO] [1686043697.455334917] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1686043697.457648273] [camera.camera]: Close Sensor - Done. 
[realsense2_camera_node-1] [INFO] [1686043697.457683973] [camera.camera]: Stop Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1686043697.457694595] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1686043697.469977227] [camera.camera]: Close Sensor - Done. 
**Some error but node can be set up.**

MartyG-RealSense, 2023-06-06:
Hi @TW050203  The **auto_exposure_limit** and **auto_gain_limit** errors occur when using firmware 5.14.0.0 with a librealsense version that is older than 2.53.1.  As you are using librealsense 2.51.1, the errors should disappear if you downgrade your D435i's firmware to the previous version **5.13.0.50**

MartyG-RealSense, 2023-06-19:
Case closed due to no further contents received.

closed 2023-06-19 by MartyG-RealSense