realsenseai/librealsense #11881: Cannot reconnect to Realsense D435 once unplugged and plugged again

mynickmynick, 2023-06-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 

| Operating System & Version |   {Win10 | 
                              | 
| Platform                                 | PC Dell  |

| Language                            |  {C++ }                          | 

### Issue Description

I have implemented a state machine to connect and stream and process frames from my realsense d435 camera. If i start the application with the camera connected everything goes well. When I unplug the camera I get errors as expected but when I reconnect I keep getting exceptions/errors and cannpot reset the connection and stream. I have tried several combinations of comands and sleeps to recover, none seems to work and always leads to the following cyclic error messages

I will paste here before my log and after a C++ code extract with the status machine, thank you very much for help

Status 0 -> 20
Succesfully set Emitter Laser
Succesfully set Emitter Laser max power = 360

 ------------- GRAB+MEDIAN TIME: 0.088 seconds.

 ------------- SEGMENTATION: 0.272 seconds.
Found a nr of Cluster: 6

 --CLUSTERING LAST PART + optional HULLS TIME: 0.006 seconds.

 ------------- -------------------- INERTIA TIME: 1.665 msec.
Loaded cloud width*height = 266142 w,h= 266142 , 1
cloud is_dense= 0
cloud isOrganized= 0

 ------------- GRAB+MEDIAN TIME: 0.078 seconds.

 ------------- SEGMENTATION: 0.251 seconds.
Found a nr of Cluster: 5

 --CLUSTERING LAST PART + optional HULLS TIME: 0.005 seconds.

 ------------- -------------------- INERTIA TIME: 1.772 msec.
Loaded cloud width*height = 266542 w,h= 266542 , 1
cloud is_dense= 0
cloud isOrganized= 0

.........

<<<<<<<  HERE I UNPLUG THE DEVICE    >>>>>>>>>>>>>>>

wait_for_frames Exception Device disconnected. Failed to recconect: No device connected8000
Status 0 -> 20
Exception-- MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "The system cannot find the path specified."
Realsense not found
Status 0 -> 20
Exception-- MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "The system cannot find the path specified."
Realsense not found
Status 0 -> 20

<<<<<<<<<  HERE I PLUG AGAIN THE DEVICE    >>>>>>>>>>>>>>>

Exception--
Failed to resolve the request:
        Format: Z16, width: 1280, height: 720

Into:
Device is already streaming!
Realsense not found
Status 0 -> 20
Exception--
[... 378 lines cut ...]
			cout << "cloud is_dense= " << _cloud->is_dense << std::endl;
			cout << "cloud isOrganized= " << _cloud->isOrganized() << std::endl;
//#endif

			if (0==i%globalState.SaveOnFile3DGrabsPeriod.get() && globalState.saveOnFile3D.get())// && !globalState.loadFromFile3D.get())
			{
				cout << "saving in ascii format \n";
				pcl::io::savePCDFile<pcl::PointXYZRGB>( "pclRGB_captured_"+to_string(i%100)+".pcd", *_cloud, false);
				pcl::io::savePCDFile<pcl::PointXYZRGB>("pclRGB_captured_average" + to_string(i % 100) + ".pcd", *_cloudAverage, false);
			}

			this_thread::sleep_for(100ms);

		}

	}
	/*catch (exception e)
	{
		cout<<"Exception "<<e.what()<<endl;//to be put in a log
		return(1);
	}
	*/

	return(0);
}

MartyG-RealSense, 2023-06-06:
Hi @mynickmynick  It's great to hear that you achieved a solution.  I will provide the feedback that I had already written.

The impression that I got from the sequence of events is that when the camera was disconnected, the pipeline may not have been closing down like it is supposed to and so when the camera is reconnected, it could not be accessed because the pipeline was still active from the first session.  If a camera is already busy when a pipe start request is made then an error such as **Failed to set power state** can occur.  

The **Failed to resolve the request** error would also occur.  This error happens when the camera cannot provide a particular stream configuration.  In this case it may be because the sensor had already been 'claimed' by the previous session and so would not become available again until the pipeline is closed.

You can use an instruction called **set_devices_changed_callback** to detect USB disconnection and reconnection events and define actions to occur when those conditions are detected, such as a camera reset.  An example of using this instruction in C++ can be found at https://github.com/IntelRealSense/librealsense/issues/9287

mynickmynick, 2023-06-07:
ok thanks I'll have a look

closed 2023-06-07 by mynickmynick