realsenseai/librealsense #11883: No RealSense devices were found! Error in docker container

kenloumixx, 2023-06-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.90                                       | 
| Platform                                 | Jetson TX2  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                          | 
| Segment			|           Robot               |  

### Issue Description
<Describe your issue / question / feature request / etc..>

I run the following command 
`ros2 run realsense2_camera realsense2_camera_node`
and the output is like below

MartyG-RealSense, 2023-06-07:
Hi @kenloumixx  Which method did you use to install librealsense and the RealSense ROS wrapper on your Jetson TX2, please?

On a Jetson board it is recommendable to install both from source code instead of using packages, building librealsense first and then the ROS wrapper secondly afterwards once it has been confirmed that librealsense is working successfully.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 has a guide for building librealsense and pyrealsense2 at the same time from source code on Jetson.

kenloumixx, 2023-06-07:
@MartyG-RealSense Thank you for your response! I followed[ the manual that you added](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049) and got this result. I also installed ROS wrapper following [this link.](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049)

MartyG-RealSense, 2023-06-07:
Thanks very much for the information.  Which RealSense camera model are you using, please?  On some models, unplugging the micro-sized end of the USB cable, turning the connector around the opposite way and re-inserting it will enable the camera to work correctly (USB-C cables are two-way insertion at the micro-sized end)

kenloumixx, 2023-06-07:
I am using depth camera D435. Since I have camera in pre-built robot, it is hard to dissect the components to unplug the camera. By the way I checked that camera works well by running the application of the robot whose mechanism I don't understand yet...

MartyG-RealSense, 2023-06-07:
Whilst a RealSense camera that is integrated into a product by its manufacturer may be similar to the standalone RealSense camera hardware, that product may use SDK software or firmware drivers that are unique to the product and not compatible with Intel's own SDK and drivers whilst working with the manufacturer's own software.  

Intel cannot provide RealSense technical support for such products unfortunately that were not manufactured by Intel and the product's manufacturer should be contacted for support instead (in this case, the company who created the robot).

kenloumixx, 2023-06-07:
@MartyG-RealSense Thank you for the response. As the [product manager](https://github.com/unitreerobotics/unitree_ros_to_real/issues/51#issuecomment-1487899083) says I can follow the Intel manuscript, can you please suggest any other ways that I can try to solve this issue?

MartyG-RealSense, 2023-06-07:
Whilst it may be possible to use the RealSense ROS wrapper on a Unitree robot, Unitree's support site will be the appropriate place to post a support question about their robots as the robot is not an Intel-made RealSense product.  

https://unitreerobotics.net/support/

Also, advice about solving problems with the RealSense ROS wrapper on computers with a USB-connected RealSense camera may not be applicable to a product with an integrated camera.

If lsusb cannot find the camera then it is likely that the problem is not specific to the ROS wrapper but is caused by something else outside of the wrapper.  So Unitree's support will be best able to diagnose that.

kenloumixx, 2023-06-07:
@MartyG-RealSense Okay, thank you for the reply and I will contact to Unitree support.

kenloumixx, 2023-06-08:
Hello, can I ask another question? 

the first line of the output of command `lsusb` suggests that it recognizes the camera. 

```
unitree@nx:~/RobotVisionSystem$ lsusb
Bus 002 Device 003: ID 8086:0b3a Intel Corp. 
Bus 002 Device 002: ID 0424:5744 Standard Microsystems Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0d8c:0012 C-Media Electronics, Inc. 
Bus 001 Device 005: ID 0424:2740 Standard Microsystems Corp. 
Bus 001 Device 008: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 007: ID 03f0:1b4a Hewlett-Packard 
Bus 001 Device 006: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 0424:2744 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

And the output of `rs-enumerate-devices` is as below 
```
unitree@nx:~/RobotVisionSystem$ rs-enumerate-devices
 08/06 13:27:07,445 ERROR [548489552192] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
Could not create device - failed to set power state . Check SDK logs for details

```
I think this is rather related to the Intel issue, rather than the issue of unitree's integrated robot product. 

If it doesn't bother you, I would like to ask how to deal with this issue.

MartyG-RealSense, 2023-06-08:
The rs-enumerate-devices message suggests that the camera cannot be accessed because it is already enabled and therefore 'busy'.

You could try resetting the camera at ROS launch with the **initial_reset** parameter.

**ros2 run realsense2_camera realsense2_camera_node  --ros-args -p initial_reset:=true**

kenloumixx, 2023-06-08:
Thank you for your response :) 

I tried both ways, but the respective answers are as below: 
```
❯ ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true
[INFO] [1686220777.229367042] [camera]: RealSense ROS v4.51.1
[INFO] [1686220777.229766785] [camera]: Built with LibRealSense v2.53.1
[INFO] [1686220777.229856545] [camera]: Running with LibRealSense v2.53.1
 08/06 10:39:37,366 WARNING [547104991552] (handle-libusb.h:109) failed to claim usb interface, interface 0, is busy - retrying...
 08/06 10:39:37,383 WARNING [547104991552] (handle-libusb.h:109) failed to claim usb interface, interface 0, is busy - retrying...
 08/06 10:39:37,395 ERROR [547104991552] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 08/06 10:39:37,395 ERROR [547130169664] (sensor.cpp:664) acquire_power failed: failed to set power state
 08/06 10:39:39,366 WARNING [547130169664] (rs.cpp:310) null pointer passed for argument "device"
[WARN] [1686220779.366683502] [camera]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1686220779.367022286] [camera]: The requested device with  is NOT found. Will Try again.
 08/06 10:39:45,435 WARNING [547113384256] (handle-libusb.h:109) failed to claim usb interface, interface 0, is busy - retrying...
 08/06 10:39:45,445 WARNING [547113384256] (handle-libusb.h:109) failed to claim usb interface, interface 0, is busy - retrying...
 08/06 10:39:45,456 ERROR [547113384256] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 08/06 10:39:45,456 ERROR [547130169664] (sensor.cpp:664) acquire_power failed: failed to set power state
 08/06 10:39:47,447 WARNING [547130169664] (rs.cpp:310) null pointer passed for argument "device"
[WARN] [1686220787.447483350] [camera]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1686220787.447677430] [camera]: The requested device with  is NOT found. Will Try again.

```

```
 ros2 launch realsense2_camera rs_launch.py initial_reset:=true
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-06-08-10-40-07-388951-nx-2704
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [2707]
[realsense2_camera_node-1] [INFO] [1686220808.063164011] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1686220808.063558443] [camera.camera]: Built with LibRealSense v2.53.1
[realsense2_camera_node-1] [INFO] [1686220808.063646667] [camera.camera]: Running with LibRealSense v2.53.1
[realsense2_camera_node-1]  08/06 10:40:08,176 WARNING [547295992128] (handle-libusb.h:109) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  08/06 10:40:08,186 WARNING [547295992128] (handle-libusb.h:109) failed to claim usb interface, interface 0, is busy - retrying...
[realsense2_camera_node-1]  08/06 10:40:08,196 ERROR [547295992128] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[realsense2_camera_node-1]  08/06 10:40:08,198 ERROR [547321170240] (sensor.cpp:664) acquire_power failed: failed to set power state
[realsense2_camera_node-1]  08/06 10:40:10,168 WARNING [547321170240] (rs.cpp:310) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1686220810.168778581] [camera.camera]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-1] [ERROR] [1686220810.169103508] [camera.camera]: The requested device with  is NOT found. Will Try again.
```
Thank you for your support..!

MartyG-RealSense, 2023-06-08:
When you provided links to how you installed librealsense and the ROS wrapper, you provided the librealsense link https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 again for ROS.  Can you confirm the method that you used to install the ROS wrapper please?  It should have been built from source code with **colcon build**.

kenloumixx, 2023-06-10:
@MartyG-RealSense Thank you for the response, yes I also installed ROS wrapper following [this link](https://github.com/IntelRealSense/realsense-ros#option-2-install-from-source) and forgot to attach. RS2_USB_STATUS_BUSY <- I think this error is related to the multiple processes that use the realsense camera. May I ask it is a common issue?

MartyG-RealSense, 2023-06-10:
When using multiple processes, access to the streams of a particular camera is governed by the rules of the **Multi-Streaming Model** at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

More than one process can access the camera, but each stream (or 'endpoint') can only be accessed by one process at a time.  so if Process 1 accesses depth then Process 2 will have access to infrared and color but not depth, as Process 1 has already claimed access rights to it.  

Once Process 1 stops using the claimed stream then it becomes available for other processes to use.

MartyG-RealSense, 2023-06-22:
Case closed due to no further comments received.

closed 2023-06-22 by MartyG-RealSense

Jellal-17, 2024-10-10:
> When using multiple processes, access to the streams of a particular camera is governed by the rules of the **Multi-Streaming Model** at the link below. [...]

Hello, I have had the same issue with the Unitree A1 robot but I killed a process that was accessing the stream, it was working then. When I rebooted the system, the first thing I did is kill the process and made sure that it wasnt running, but then I tried to use the `realsense-viewer`,  there was no camera. So, I uninstalled the SDK and installed it fresh. It was the same loop again, works till reboot and doesnt work after that no matter what. the result of `rs-enumerate-devices` is
No device detected. Is it plugged in?