realsenseai/librealsense #11885: In the rs server function, deep video streams and rgb video streams are not obtained through the viewer

H-chun, 2023-06-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | 2.53.1 | 
| Operating System & Version |  jetson5.1.1 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.53.1 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Currently, I want to obtain each frame of the depth map and each frame of the rgb simultaneously through RS server. Is there any method available

MartyG-RealSense, 2023-06-07:
Hi @H-chun  Using a D457 with rs-server would be problematic.  The D457 in MIPI mode is not supported by rs-server.  And if the camera is used in USB mode and detected as D455, rs-server does not have full support for D455 because updates of the tool ceased before D455 was released.

A RealSense user recently demonstrated at https://github.com/IntelRealSense/librealsense/issues/11857 that D455 could be used with rs-server.  However, the tool will not support use of the D455's 5 FPS minimum speed mode and D455 does not support '6' as a minimum FPS , so you would be unable to use the 1280x720 at 6 FPS modes listed at the link below.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#36-software-limitations

H-chun, 2023-06-08:
Hello, I have understood everything you mentioned. Currently, what I would like to inquire about is how to obtain the data of each compressed image frame transmitted in RS server? Is there a specific function or are there examples to obtain? Do you have any suggestions? thanks

MartyG-RealSense, 2023-06-08:
You can access the data transmitted by rs-server using the RealSense Viewer on the 'host' computer (the one that does not have the camera attached).  This is done by setting up a Network Camera using the **Add Source** button at the top of the Viewer's options side-panel.

You can also access the networked data on the host without using the Viewer by creating a program script,  A C++ example of doing so can be found at https://github.com/IntelRealSense/librealsense/issues/9376

closed 2023-06-21 by MartyG-RealSense