realsenseai/librealsense #11888: Question: Can 2 different applications consume Depth and Color independently using Real Sense SDK.

rajrajeswaran, 2023-06-07:
Camera Model: D415
Firmware Version: 05.12.07.100
Operating System & Version: Windows 10 Enterprise 1809                    | 
SDK Version: 2.38.1
Language: C#

We are considering building 2 separate C# applications.  One will be designed to consume Depth data from the Intel RealSense D415 while other will consume RGB/IR.  Both of these application will be running simultaneously processing streams for 2 different use cases.  Is this possible with Intel Real Sense SDK?  Or only one process can be attached to a camera at a time?  In device manager, the Depth and RGB are listed as separate hardware so I am wondering if a process can simply attach to Depth and another process attach to RGB.

MartyG-RealSense, 2023-06-08:
Hi @rajrajeswaran  The rules of the **Multi-Streaming Model** of the RealSense SDK allow for more than one process to use the same camera so long as each stream type ('endpoint') is only accessed once.  So if one process accesses Depth and the other process accesses RGB and IR then that should be fine.  The second process would not be able to access Depth as it would be claimed by the first process.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

rajrajeswaran, 2023-06-09:
@MartyG-RealSense As always thank you for the quick turn around.  Appreciate the information.  I will close this.

closed 2023-06-09 by rajrajeswaran

MartyG-RealSense, 2023-06-09:
You are very welcome.  As always, I am pleased to be able to help.  :)