realsenseai/librealsense #11890: T265 does not appear in the realsense-viewer

pat19218, 2023-06-07:
| | |
|---|---|
|**librealsense**|2.54.1 RELEASE|
|**OS**|Windows|

I connect and disconnect the t265 camera and it is not listed in the realsense-viewer program. I have pyrealsense installed and when executing a script that extracts information from the camera if it does it correctly, I imagine it is a realsense viewer error.

tugepaopaoo, 2023-06-08:
This version no longer supports T265, you need to download the previous version， such as v2.53.1.
[v2.54.1 Release Note](https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1)

pat19218, 2023-06-08:
> This version no longer supports T265, you need to download the previous version， such as v2.53.1. [v2.54.1 Release Note](https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1)

Thanks!

closed 2023-06-08 by pat19218

buicongnguyen80, 2023-12-19:
git clone --single-branch -branch R/253 https://github.com/IntelRealSense/librealsense.git
I do as this guide but when I connect T265 to RB5 device and run realsense-viewer. I receive message: 
There are 0 connected RealSense devices.
What should I do to have T265 displayed on realsense-viewer ?

ConnorAtmos, 2024-06-11:
Here is my respective installation steps to install the realsense version that works with T265:

sudo apt-get install librealsense2=2.53.1-0~realsense0.8251

sudo apt-get install librealsense2-gl=2.53.1-0~realsense0.8251

sudo apt-get install librealsense2-net=2.53.1-0~realsense0.8251

sudo apt install librealsense2-dkms=1.3.22-0ubuntu1

sudo apt install librealsense2-utils=2.53.1-0~realsense0.8251