realsenseai/librealsense #11891: change the exposure and save?

luobodan, 2023-06-08:
hello! 
I want to change the stereo exposure from "auto" to "Manual setting".  I just find it with changing the value of "exposure" in realsense-viewer , but when I reboot the device,  the value is back .How to save it on the chip and don't need to change it every time  open the device?

luobodan, 2023-06-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d435i } | 
| Firmware Version                   | (Open RealSense Viewer --> v2.50.0) | 
| Operating System & Version |   {ubuntu18.04  | 
| Kernel Version (Linux Only)    |  (5.4.0-139-generic)                                         | 
| Platform                                 | PC-  |
| SDK Version                            |  { legacy / 2.50.0 }                          | 
| Language                            |  {c++ }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
hello!
I want to change the stereo exposure from "auto" to "Manual setting". I just find it with changing the value of "exposure" in realsense-viewer , but when I reboot the device, the value is back .How to save it on the chip and don't need to change it every time open the device?

MartyG-RealSense, 2023-06-08:
Hi @luobodan  Except for calibration values, the values of settings cannot be saved to the camera hardware.

If your realsense-viewer is not preserving settings inbetween sessions then you could save a **json** configuration file using the export icon near the top of the options side-panel and load the file back in at your next session.

[image]

If you are only changing exposure though then it may be faster to just continue changing it manually.

The settings loss does not seem to occur with the Windows version of realsense-viewer, with the manual exposure value preserved between sessions even after the computer is rebooted.

MartyG-RealSense, 2023-06-21:
Case closed due to no further comments received.

closed 2023-06-21 by MartyG-RealSense

luobodan, 2023-07-05:
thank you very much!!!