realsenseai/librealsense #11893: Weird Looking Depths on Unreal Engine 5.1 with C API

ffreality, 2023-06-08:
| Camera Model                       |D455| 
| Firmware Version                   |5.15.02| 
| Operating System & Version | Windows 11| 
| Platform                                 |PC|
| SDK Version                            |2.54.1| 
| Language                            |C and Unreal Engine| 
| Segment			|VR| 

### Issue Description
I created an Unreal Engine 5.1 / 5.2 plugin with v2.54.1 for Windows and Android platforms. (I didn't use UE wrapper. I re-wrote it)
Right now RGB stream works perfectly with both platforms but depth streams looks like this.

1- Using colorizer
[image]

```
rs2_frame* Rs_Frames = rs2_pipeline_wait_for_frames(In_Device->Rs_Pipeline, Timeout, NULL);
rs2_frame* First_Frame = rs2_extract_frame(Rs_Frames, 0, NULL);

rs2_processing_block* Colorizer = rs2_create_colorizer(NULL);
rs2_frame_queue* frame_queue = rs2_create_frame_queue(1, NULL);
rs2_start_processing_queue(Colorizer, frame_queue, NULL);
rs2_frame_add_ref(First_Frame, NULL);
rs2_process_frame(Colorizer, First_Frame, NULL);

rs2_frame* colorized_frame = rs2_wait_for_frame(frame_queue, Timeout, NULL);

Stream Type = RS2_STREAM_DEPTH
RS2 Format = RS2_FORMAT_Z16

and I create UTexture2D for depth with this
Out_T2D = UTexture2D::CreateTransient(Size.X, Size.Y, PF_FloatRGB)
```

2- If I don't use colorizer and just use ``First Frame`` to create UTexture2D it look like this.
[image]

NOTE:
I am using C API because Unreal Engine doesn't start with C++ API. It gives DLL delayed load error.
I added Intel.Realsense.dll, realsense2.dll and realsense2.lib and I know how can I use delayed load (same structure works with C API and I created lots of plugins, before including PDFium, libharu and NDI)

ffreality, 2023-06-08:
Using `rs2_create_sync_processing_block` gives better result but I need to help about two things.
1- How can I get only one eye ?
2- This guide and some issues reported that exposure is important for depth. How can I change it with C API ? for example how can initialize rs2_options ?

MartyG-RealSense, 2023-06-08:
As the displayed image is a depth image, it looks as though there should be only one image but it is being duplicated.  I don't have a suggestion for why that might be occurring.

There are few C-language script references available unfortunately, but one at https://github.com/IntelRealSense/librealsense/issues/7817#issuecomment-730699218 should be helpful for writing a set_option instruction.

Set RS2_OPTION_ENABLE_AUTO_EXPOSURE to 0 to disable it and 1 to enable it.

To set a manual depth exposure value after disabling auto-exposure, use something like:

`depth.set_option(RS2_OPTION_EXPOSURE, 7000);`

ffreality, 2023-06-09:
Find the problem.
My colorizer function is true but I think created result `R8G8B8 `but Unreal doesn't have that kind of pixel format. We have `B8G8R8A8 `or `R8B8G8A8`

For this reason, passing Z16 format directly to a `PF_ShadowDepth` without colorizer can show visually correct texture but there is no alpha. black areas are just black. So, I can't use them for masking.

Probably I need to create a temp texture as local variable, read pixel values as color, re-arange colors and update current texture with it.

MartyG-RealSense, 2023-06-09:
Thanks so much for the update of your progress!

ffreality, 2023-06-09:
Okay finished. Only thing I need is solving exposure options with C.

closed 2023-06-09 by ffreality

MartyG-RealSense, 2023-06-10:
Great news!  Thanks so much again for creating this plugin  :)

ranjitkathiriya544, 2023-09-01:
Where is the link for the latest Unreal engine 5 plugin? or github repo

ffreality, 2023-09-01:
> Where is the link for the latest Unreal engine 5 plugin? or github repo

I personally developed it and don't consider to share it right now.