realsenseai/librealsense #11895: Python Wrapper pipeline.start() Segmentation Fault

cu-boulder-robotics, 2023-06-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.55 (Built from source) | 
| Operating System & Version | Ubuntu 20.04.6 | 
| Kernel Version (Linux Only)    | 5.10.66 | 
| Platform                                 | Khadas Edge2 (ARM 64) |
| SDK Version                            | 2.54.1 | 
| Language                            | python | 
| Segment			| Robot/others | 

I am attempting to use the pyrealsense2 library to capture frames from a single D455 on a Khadas Edge2. Since ARM is not officially supported, I built from source using the following cmake flags:

`cmake …/ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND:bool=true -DBUILD_WITH_CUDA:bool=false -DBUILD_GRAPHICAL_EXAMPLES:bool=false -DCMAKE_BUILD_TYPE=release`

The realsense viewer works fine for both depth and color, and so do the example CPP scripts. However, using python causes a segmentation fault when calling `pyrealsense2.pipeline().start()`.

Python script that recreates the issue:
```python3
import pyrealsense2 as rs
pipe = rs.pipeline()
pipe.start() # Segmentation fault
```

This appears to be a problem with the python wrapper, given that the CPP `pipe.start()` function works just fine.

MartyG-RealSense, 2023-06-09:
hi @cu-boulder-robotics  Using firmware version 5.13.0.50 may cause errors when used with librealsense SDK version 2.54.1.  This is because it was compatible with SDKs up to 2.51.1 but not 2.53.1.  Please update your firmware to **5.15.0.2**, the recommended firmware version for 2.54.1.  You can perform this firmware update in realsense-viewer in the drop-down menu of the **More** option near the top of the realsense-viewer options side-panel.

closed 2023-06-13 by cu-boulder-robotics

MartyG-RealSense, 2023-06-14:
That's excellent to hear that a firmware update resolved your problem.  Thanks very much for the update!