realsenseai/librealsense #11897: Problems building and installing on M1 Mac

emilk, 2023-06-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |  | 
| Operating System & Version |  MacOS 13.3.1 (a)   | 
| SDK Version                            |                            | 
| Language                            |  C / Python                          | 

### Issue Description
I'm trying to get a RealSense D435 working on my MacBook Pro M1. 

I tried building from source following the instructions on https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

I get this:

```
§ cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- The C compiler identification is AppleClang 14.0.3.14030022
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.54.1
-- Setting Unix configurations
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Failed
CMake Error at CMake/lrs_macros.cmake:17 (message):
  Project 'realsense2' requires C++14 or higher
Call Stack (most recent call first):
  CMakeLists.txt:53 (config_cxx_flags)
```

### Python
I want to use it from either Python or C. **Is there a pip wheel I can just install and use?**

I tried https://pypi.org/project/pyrealsense2-macosx/ but I get:

```
profile = pipe.start()
RuntimeError: No device connected
```

despite the camera being connected (and working correctly as a web-cam)

MartyG-RealSense, 2023-06-09:
Hi @emilk  There is a step by step guide for installing the librealsense SDK on an Apple Silicon M1 Mac at the link below.

https://lightbuzz.com/realsense-macos/

Alternatively, the **brew install librealsense** installation method also supports Apple Silicon.

https://formulae.brew.sh/formula/librealsense

MartyG-RealSense, 2023-06-16:
Hi @emilk  Do you require further assistance with this case, please?  Thanks!

emilk, 2023-06-19:
I tried `brew install librealsense` but I still fail to find the camera using something like `pyrealsense2`.

I also tried [a Rust wrapper around the C library](https://github.com/Tangram-Vision/realsense-rust/pull/16) but `rs2_create_device` returns an `rs2_error` of `8`, which is weird, since there are only eight errors, 0-7: https://intelrealsense.github.io/librealsense/doxygen/rs__types_8h.html#ab73d6772c40d1a2fba645a1d0a7eed5e

emilk, 2023-06-20:
Yup, I'm on Ventura

MartyG-RealSense, 2023-06-20:
https://github.com/IntelRealSense/librealsense/issues/11815 is the central place on this support forum where I am directing Ventura users who are having problems so that they can discuss it amongst themselves, and you will therefore be notified automatically of any solutions that the RealSense Mac community comes up with for getting librealsense working on Ventura.

MartyG-RealSense, 2023-07-03:
Hi @emilk  Bearing in mind the advice above about posting a comment on https://github.com/IntelRealSense/librealsense/issues/11815 - do you require further assistance with this case, please?  Thanks!

MartyG-RealSense, 2023-07-09:
Case closed due to no further comments received.

closed 2023-07-09 by MartyG-RealSense

emilk, 2023-07-10:
Sorry for the slow reply from me - I've given up on using RealSense on Mac for now.