realsenseai/librealsense #11904: Help needed to improve glare in night-time images captured with artificial lighting

DaphnaNanovel, 2023-06-12:
Camera Information:
1. Camera Model: D405

System Information:
1. Operating System & Version: Ubuntu 18.04
2. Platform: NVIDIA Jetson
3. Language: Python
4. Segment: Robot

Description:
I am encountering an issue with glare in images captured during nighttime using artificial lighting sources. The presence of glare in these images hampers visibility and affects the overall quality. I am seeking guidance and assistance on techniques or approaches to effectively reduce or eliminate glare from such images.

Background:
1. The images are taken at night using artificial lighting, resulting in bright spots and reflections.
2. The glare is causing a loss of detail and impacting the image quality.

Issue:
I would greatly appreciate any advice, suggestions, or pointers to relevant resources on how to effectively address and reduce glare in night-time images captured with artificial lighting using the D405 (open frame)

Thank you for your support and guidance.
[image]

MartyG-RealSense, 2023-06-18:
Hi @DaphnaNanovel  Do you require further assistance with this case, please?  Thanks!

DaphnaNanovel, 2023-06-18:
Hi @MartyG-RealSense, I would appreciate your guidance on where to find a suitable filter for RealSense 405 (open frame) to mitigate the glare. Any recommendations on retailers or suppliers would be greatly appreciated. Thank you!

MartyG-RealSense, 2023-06-18:
There is not a specific recommendation for suppliers (as polarization filter that is linear will work except for the circular ones in 3D glasses).  The filter supplier at the link below has existed for a long time and should therefore be reliable.

https://www.edmundoptics.com/f/linear-glass-polarizing-filters/11758/

closed 2023-06-18 by DaphnaNanovel