realsenseai/librealsense #11910: Sensor streaming issue and Enabling Librealsense Logs Unity

KIL299, 2023-06-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   {Win (11) }  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {c#/unity }                          | 
| Segment			|  { }                          | 

### Issue Description
I have a question.
I am running L515 sensors using Realsense unity package. When I start and stream through a particular sensor multiple times, the sensor is not able to receive frames after some time.
What I am doing is 
1. start a sensor
2. start streaming
3. stop sensor

when repeated multiple times after some iterations no frames are being received from sensor. 
And the interesting thing I am noticing is when I start the above process again the same sensors starts receiving frames.

So I wanted to check the realsense logs generated for sensors to debug. 
but unable to find a way to get those.

I tried the below link to enable the logs but unable to find the logs. And still don't have any idea if the logs are being generated or not. 
https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-how-do-i-enable-librealsense-logs

MartyG-RealSense, 2023-06-14:
Hi @KIL299  The path of the folder on Windows where logs are stored is **C: > Users > <username> Documents**

For example, if your Windows username was KIL299 then the path to the log file location would be `C:\Users\KIL299\Documents`

KIL299, 2023-06-21:
Hi @MartyG-RealSense 
I have a attached a log file below 
here's what I am doing
1. Starting 8 sensors 
2. starting stream for those 8 sensors
3. stopping those 8 sensors

repeat the above steps multiple times 

In my case it runs for about 3-4 times then suddenly we couldn't receive frame points from device
[LibrealsenseLogs_.txt.zip](https://github.com/IntelRealSense/librealsense/files/11817770/LibrealsenseLogs_.txt.zip)

MartyG-RealSense, 2023-06-21:
If you are enabling 8 L515 cameras simultaneously then doing so without problems would require a powerful computer, as the more cameras that are active simultaneously, the more of the computer's resources that are consumed.  For just 4 cameras, a computer with a processor chip equal to Intel Core i7 is recommendable.

A way around this processing burden is to use **hardware sync**.  When used with the L515 camera model, it cycles through the multiple attached cameras, activating capture at different instants so that all cameras are not active simultaneously.  Intel provide a guide about this at the link below.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

In Unity, hardware sync can be enabled by adding a RealSense wrapper component called **RsDeviceInspector** to your RsDevice object using the 'Add Component' button at the bottom of the Inspector and typing 'rsdeviceinspector' into the Add Component interface's search box.  

The settings of RsDeviceInspector are initially hidden but once a scene is played and the camera begins streaming, the settings become accessible and a tick can be placed in a tick-box beside the **InterCamSyncMode** option to enable that camera as a hardware-synced one.

[image]

Alternatively, you could use a more powerful computer to enable 8 cameras simultaneously without using hardware sync.  A processot equivalent to an **i9** may be able to handle this processing burden.

KIL299, 2023-06-22:
@MartyG-RealSense 
we are currently using processor chip equal to Intel Core i7. 
will try the intercamsync mode as you suggested and will get back to you.

MartyG-RealSense, 2023-06-22:
Thanks very much for the update.  I look forward to your next report.  Good luck!

KIL299, 2023-06-26:
Hi @MartyG-RealSense 
I have uploaded below file that was generated when device was not able to capture streams on starting streams. 
and i would like to specify that I have more than 3 setups available. 
out of these setups I am getting the issue in one of the setup. 
Do you have any idea that why would this happen?
Only one particular sensor is causing the issue. 

[RealSenseLogs.txt](https://github.com/IntelRealSense/librealsense/files/11868756/RealSenseLogs.txt)

MartyG-RealSense, 2023-06-26:
Have you tried unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  A characteristic of the L515 camera model is that it works best when the connector is in one particular orientation of the two insertion directions available.

MartyG-RealSense, 2023-06-26:
Regarding the error: `GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."` - this is a highly rare error that has only appeared a couple of times in past cases, and none with L515.

It is likely to be an issue with the camera rather than the computer or the SDK installation, as your other L515 work with it.

You could try downgrading your L515's firmware version from 1.5.8.1 to **1.5.5.0** to eliminate the possibility that the firmware driver is damaged on that particular camera.

https://dev.intelrealsense.com/docs/firmware-releases-l500

KIL299, 2023-06-26:
Okay will look into and get back to you.

MartyG-RealSense, 2023-07-03:
Hi @KIL299  Do you have an update about this case that you can provide, please?  Thanks!

KIL299, 2023-07-03:
Hi @MartyG-RealSense 
I tried the connection disconnection but not improvements were found
so need to check by swapping the sensors
Will get back to you once it's tested.

MartyG-RealSense, 2023-07-17:
Yes, it is totally fine to close this case and return to it at a future date if you need to.