realsenseai/librealsense #11914: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

FeiSong123, 2023-06-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |01.05.08.01 | 
| Operating System & Version | Ubuntu20.04  | 
| Kernel Version (Linux Only)    | 5.15.0-73-generic                                        | 
| Platform                                 | PC.  |
| SDK Version                            |  2.53.1                       | 
| Language                            | C++                          | 
| Segment			|  Robot                     | 

### Issue Description
<Describe your issue / question / feature request / etc..>

terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

MartyG-RealSense, 2023-06-14:
Hi @FeiSong123  A **xioctl(UVCIOC_CTRL_QUERY)** error typically indicates that there is a conflict with the kernel.  You are using a supported kernel version (5.15) though.

The **Device or resource busy** type of xioctl error suggests that the camera might be inaccessible because it has already been started and so is 'busy'.  This might occur if a previous running of a script was not properly shut down and so the camera pipeline is still open on the subsequent run because it was not closed when the previous run of the script was stopped.

Does the error occur when the script is run for the first time after having booted up the computer, please?

FeiSong123, 2023-06-14:
I often get this error when I run the script for the first time after starting up

MartyG-RealSense, 2023-06-14:
Please try unplugging the micro-sized end of the USB cable from the base of the L515 camera, turning the connector around the other way and plugging it back into the camera just to confirm that the problem is not related to the direction of the connector.  This is because the L515 model can work better when the micro-sized connector is inserted in one particular orientation of the two possible directions available.

FeiSong123, 2023-06-15:
> This might occur if a previous running of a script was not properly shut down and so the camera pipeline is still open on the subsequent run because it was not closed when the previous run of the script was stopped.

Is there any instruction to completely close the camera pipe line, so that as long as I completely close the camera pipeline every time I open the program, this error should not happen again

MartyG-RealSense, 2023-06-15:
You could do **pipe.stop()** and then **pipe.close()** on the next line.

Instead of stopping the pipeline, **sensor.stop()** and **sensor.close()** to stop and close the sensor can also be used if pipeline stop is causing problems, as described at https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103

FeiSong123, 2023-06-17:
Are there command that run outside the terminal instead of inside the program

MartyG-RealSense, 2023-06-17:
It is possible to reset the entire USB port instead of the camera specifically by using an Ubuntu **bash** script like the example of one at the link below.

https://github.com/IntelRealSense/librealsense/issues/8393

MartyG-RealSense, 2023-06-25:
Hi @FeiSong123   Do you require further assistance with this case, please?  Thanks!

FeiSong123, 2023-06-25:
With your help I have solved this problem, thanks! You can close this topic at any time.

MartyG-RealSense, 2023-06-25:
That's excellent news that you succeeded.  Thanks very much for the update.  As you suggest, I will close the topic.  Thanks again!

closed 2023-06-25 by MartyG-RealSense