realsenseai/librealsense #11916: Program Error

Nandomonaiya, 2023-06-15:
| | |
|---|---|
|**librealsense**|2.54.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D455|
|**Serial Number**|234222304426|
|**Firmware Version**|5.15.0.2|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|229123063074|
|**Firmware Update Id**|229123063074|

When running the box dimensional scanner program on python, the program only works once, after the program ran for the first time, it calibrates and measure the volume of the object correctly, but when the second ran, it didn't calibrate correctly, and couldn't measure or detect anything. The issue was found after updating the firmware to the newest version. The camera must be repeatedly reconnect to the pc via usb to begin the correct measurement and calibration, every time the programs wants to be run.

The first picture is the first run with the same config:
[image]
The second picture is the second run with the same config:
[image]

MartyG-RealSense, 2023-06-15:
Hi @Nandomonaiya  Can you first try downgrading the firmware to the previous version **2.14.0.0** to confirm whether the firmware update is the cause of the problem, please?  The link below has instructions for performing the downgrade in the Viewer.

https://support.intelrealsense.com/hc/en-us/community/posts/15226169805971/comments/15228046150035

Nandomonaiya, 2023-06-16:
The issue remains the same, whenever i calibrate the camera for the first run, it detects the depth and could display it in 3D, but when i try for the 2nd run, the camera only detects the image as in 2D
[image]

MartyG-RealSense, 2023-06-23:
Hi @Nandomonaiya  Sincere apologies for the delay in responding further.  Does your situation remain the same or do you have progress to report?

closed 2023-06-30 by MartyG-RealSense