realsenseai/librealsense #11918: Docker - No RealSense devices were found

hackerclemens, 2023-06-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Ubuntu 22.04.2 LTS | 
| Kernel Version (Linux Only)    |     6.2.10-surface  | 
| Platform                                 | PC - Docker |

### Issue Description
Hey, I want to use a Realsense D435 for a project I am working on. For the project, I am running a Docker with an Ubuntu 20 image and ROS-noetic running. Unfortunately whenever I want to start the ROS-Realsense wrapper, I get the error-message, that no Realsense device has been found. 

[image]

I can connect to the camera when I am not in the docker, using the Realsense viewer. So the problem has to be somewhere in the docker configuration, but I can't find it. I made sure that the docker has access to the USB-port, by adding the /dev folder to the shared files in Docker-desktop and adding the devices in the docker-compose.yaml file. When I am in the Docker I see that there is an usb-file in /dev/bus/usb/003, which is the port the camera is connected to. 

[image]

Here is one of the services in my docker-compose.yaml file:

```
  rosclient:
    depends_on:
      - novnc
      - roscore
    image: ros1_base
    build:
      context: .
      dockerfile: Dockerfile_including_yolo
    environment:
      - DISPLAY=novnc:0.0
      - ROS_MASTER_URI=http://roscore:11311
    networks:
      - ros
    volumes:
      - "./projects:/catkin_ws/src/projects:rw"
    privileged: true
    stdin_open: true
    tty: true
    devices:
      - "dev:/dev" 
 ```

MartyG-RealSense, 2023-06-16:
Hi @hackerclemens  Could you first confirm for me which librealsense SDK and ROS wrapper versions you are using, please? 

*********

The ROS1 wrapper for Noetic should be used with SDK version 2.50.0 or 2.51.1.  The wrapper has not been updated for newer SDK versions such as 2.53.1 and 2.54.1 because development of the ROS1 wrapper has ceased and development focus is now on the ROS2 wrapper.

If you are using SDK 2.50.0 or 2.51.1 then you should be using the older firmware driver version **5.13.0.50** as newer firmwares such as 5.14.0.0 and the 5.15.0.2 that you are using are not fully compatible with 2.50.0 and 2.51.1 and can generate errors.

Furthermore, the librealsense SDK does not currently have official support for kernel 6.2.  In SDK versions 2.50.0 and 2.51.1, the most recent officially supported kernel was 5.11.  SDK version 2.53.1 added support for kernel 5.15 on Ubuntu 22.04 Jammy.  Intel have opened a feature request to investigate implementation of kernel 6.1 but it is not implemented in the SDK yet.

When using an unsupported kernel, it is recommended that librealsense be built from source code with CMake with the **-DFORCE_RSUSB_BACKEND=TRUE** build flag included so that the SDK is built in RSUSB mode, which bypasses the kernel.

I note that you have been able to access the RealSense Viewer, so your librealsense installation is working despite the kernel version.

MartyG-RealSense, 2023-06-23:
Hi @hackerclemens  Do you require further assistance with this case, please?  Thanks!

hackerclemens, 2023-06-26:
No thank you. 
I've given up and installed Ubuntu 20.04, so I did not have to use a docker. It immediately worked with the same SDK-Version and ROS-Realsense wrapper. I suspect that the problem had to do something with the Docker.

MartyG-RealSense, 2023-06-26:
Thanks very much for the update.  I'm pleased to hear that you achieved a solution.  As you do not require further assistance, I will close this case.  Thanks again!

ruthvik92, 2023-08-17:
@MartyG-RealSense : Sorry to comment on a closed issue. I just upgraded my kernel to `6.2.0.26` on my `Ubuntu 22.04`. To my surprise, `realsense-viewer` worked fine but `librealsense2` `.so` file could not be loaded according to this following error. I'm using `ROS2 Humble` 
````
ros2 launch realsense2_camera rs_launch.py camera_name:=main_camera
[INFO] [launch]: All log files can be found below /home/ruthvik/.ros/log/2023-08-17-13-15-47-405770-ruthvik-lw-17592
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [17599]
[realsense2_camera_node-1] terminate called after throwing an instance of 'class_loader::LibraryLoadException'
[realsense2_camera_node-1]   what():  Could not load library dlopen error: librealsense2.so.2.53: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
[ERROR] [realsense2_camera_node-1]: process has died [pid 17599, exit code -6, cmd '/home/ruthvik/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=main_camera -r __ns:=/main_camera --params-file /tmp/launch_params_n5nydhdi --params-file /tmp/launch_params_x24rpk8e'].
````

What is going on ? Will building from source fix the issue ?

MartyG-RealSense, 2023-08-18:
It's great to hear that you found a solution, @ruthvik92 - thanks very much for the update!