realsenseai/librealsense #11919: Cannot build realsense-ros package saying "depends on unknown jobs: librealsense2"

Karin-Sugi, 2023-06-17:
Hi, I am trying to use realsenese with RaspberryPi 4. Now, I've cloned librealsense and built it. Thus, as a next step, I cloned realsense-ros package and tried to built it. However, it is avandoned saying the packages is `depends on unknown jobs: librealsense2`.

What I did for install is following.
```
// install librealsense
$ git clone https://github.com/IntelRealSense/librealsense.git
$ cd librealsense/
$ sudo apt install -y git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
$ sudo apt install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
$ sudo apt install -y cmake
$ ./scripts/setup_udev_rules.sh
$ ./scripts/patch-realsense-ubuntu-lts.sh
$ mkdir build && cd build
$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
$ sudo make clean && make && sudo make install

// install realsense-ros package
$ cd ~/hauser_ws/src
$ sudo apt install ros-melodic-ddynamic-reconfigure
$ git clone https://github.com/IntelRealSense/realsense-ros.git
$ cd realsense-ros/
$ git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
$ cd ../../
$ catkin init
$ catkin clean
$ catkin config --cmake-args -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
$ catkin build
```

How can I fix it?

MartyG-RealSense, 2023-06-17:
Hi @Karin-Sugi  After the **git_checkout** git tag command, your list of instructions is different from the ones in the official instructions for installing the ROS1 wrapper from source code.  The remaining instructions after git checkout in the official instructions are:

```
cd ..
catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
```

For example, with your instruction **cd ../../** before init instead of the official instructions' method of cd.., init and then another cd.. I wonder if you are ending up in an incorrect folder when inputting init.

Karin-Sugi, 2023-06-18:
Hi, thanks for your reply. As you said i Tried following command, but it got another error!

```
// install realsense-ros package
$ cd ~/hauser_ws/src
$ sudo apt install ros-melodic-ddynamic-reconfigure
$ git clone https://github.com/IntelRealSense/realsense-ros.git
$ cd realsense-ros/
$ git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
$ cd ..
$ catkin init
$ cd ..
$ catkin clean
$ catkin config --cmake-args -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
$ catkin build
```
Message shown after `catkin build` is that
```
[build] Warning: Skipping package `librealsense2` because it has an unsupported package build type: `ament_cmake`
[build] Note: Avalirable build types:
[build] - `catkin`
[build] - `cmake`
Abondoned << realsense2_camera [ Depends on unknown jobs: librealsense2]
```

It looks the `realsense2_description` is built, and only `realsense2_camera` package is in trouble.

MartyG-RealSense, 2023-06-19:
Raspberry Pi is one of the most difficult computing devices to get the RealSense ROS wrapper installed successfully on, unfortunately.

A couple of suggestions that I would make are:

1.  Try to install librealsense and the ROS1 wrapper together at the same time from packages instead of separately from source code.  This can be done with the command below, though this install method will not install the RealSense Viewer tool.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

Even if the Viewer was not installed, this test would at least establish whether it was possible to get the ROS wrapper working on your Pi with this method.

2.  Alternatively, try building the wrapper following the official installation instructions instead of using your own catkin workspace path.

Karin-Sugi, 2023-06-20:
I see.
I have a question when i try installation from the package, what should i do about librealsense i already installed. Should i delete them? Or i can still use it as a viewer?

MartyG-RealSense, 2023-06-20:
I would usually recommend removing a previous installation before performing a new installation.  This can be done in the following ways.

**ROS wrapper source code build**
Delete the entire /catkin_ws/src catkin workspace directory.

**librealsense package installation**
Input the command below in the Ubuntu terminal to remove all RealSense related packages from the computer.
`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

**librealsense source code build**
Go to the **build** folder of your previous librealsense source code installation and input the CMake command below.
`sudo make uninstall && make clean`

Karin-Sugi, 2023-06-23:
Hi, I cleaned up source code and re-install `ros-melodic-realsense2-camera`.
Now, I checked all topics to find color image topic(`/camera/color/image_raw`) isn't working. When I try to see the color image using rqt, the image is grey out and error message shows up on terminal saying `control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`.

On the other hand, depth image topic (`/camera/depth/image_rect_raw`) works fine :)

How can I solve it?

MartyG-RealSense, 2023-06-23:
Does the color topic remain unavailable if you add the commands below to your roslaunch instruction?

`enable_color:=true initial_reset:=true`

closed 2023-07-08 by MartyG-RealSense