realsenseai/librealsense #11924: Bounding Box Error (not showing) in Python

Nandomonaiya, 2023-06-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
[image]
When i the box_dimensioner_multicam ran in Python, the camera doesn't detects anything within the chessboard, the calibration was succeed however there was no object detected, i must re-plug the usb cable to the PC to make the camera's system reset and the code works after the reset. That is the only fix that i've just discovered. Is it something wrong with the camera?

MartyG-RealSense, 2023-06-19:
Hi @Nandomonaiya  With the **box_dimensioner_multicam.py** example program, the chessboard must be empty of objects when the program launches and performs calibration.  You will then be prompted to place an object on the chessboard to measure it.  Can you confirm whether the chessboard has no object on it when you first run the program, please?

Nandomonaiya, 2023-06-20:
Already confirmed that there were no objects on the chessboard. I tried to run the program and make several configurations myself last week, but when i tried it on recently, the camera didn't detect the bounding box ever since, i already did the hardware reset and ran the default box dimensional multicam from the template, but the result was the same, no bounding box showed

MartyG-RealSense, 2023-06-20:
Can you confirm that the object is successfully detected and the bounding box is generated after resetting the camera with a USB uplug-replug, please?

Also, have you tried a different colored object with the camera?  It is a general physics principle (not specific to RealSense) that dark grey or black absorbs light and so makes it more difficult for depth cameras to read depth information from such surfaces. 
 The darker the color shade, the more light that is absorbed and so the less depth detail that the camera can obtain.

closed 2023-07-05 by MartyG-RealSense