realsenseai/librealsense #11927: disable link time optimization for fw project

Carl-1999, 2023-06-20:
Hi! Here are some problems I can't solve.

camera Model :   D415 
ubuntu                 20.04.6LTS，
uname -r              5.15.0-75-generic
run command : cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release 

there is the output:
Checking internet connection...
Internet connection identified
Info: REALSENSE_VERSION_STRING=2.54.1
Setting Unix configurations
Building libcurl enabled
using RS2_USE_V4L2_BACKEND
Check for updates capability added to realsense-viewer
Check for updates capability added to realsense-depth-quality
Fetching recommended firmwares:
disable link time optimization for fw project
... https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.15.0.2.bin
... https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
... https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
Configuring done
Generating done

but when I make and install this project,I can not start realsense-viewer.
The output is realsense-viewer: command not found

[image]

karinoon, 2023-06-24:
Hi @Carl-1999 Have you solved this problem?

Carl-1999, 2023-06-30:
> Hi @Carl-1999 The flag -DBUILD_EXAMPLES=TRUE will only install text-based tools and examples that do not have graphics, such as rs-hello-realsense. To include graphics support and graphical examples and tools such as realsense-viewer in the SDK build, you should also add the flag **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** to your CMake instruction.

Thank you very much for your advice！ But I run the following command：`cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release  -DBUILD_GRAPHICAL_EXAMPLES=TRUE -DCMAKE_INSTALL_PREFIX=/usr/local/librealsense
`
/home/sha/Pictures/2023-06-30 11-18-41屏幕截图.png

and I make install successfully，I still can not run realsense-viwer.

Carl-1999, 2023-06-30:
> Hi @Carl-1999 The flag -DBUILD_EXAMPLES=TRUE will only install text-based tools and examples that do not have graphics, such as rs-hello-realsense. To include graphics support and graphical examples [...]

Thank you very much for your advice！
 But I run the following command：cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DBUILD_GRAPHICAL_EXAMPLES=TRUE -DCMAKE_INSTALL_PREFIX=/usr/local/librealsense

the output is:
Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.54.1
-- Setting Unix configurations
-- Fetching pybind11...
-- pybind11 v2.10.3 
-- Fetching pybind11 - Done
-- Building libcurl enabled
-- using RS2_USE_V4L2_BACKEND
-- Check for updates capability added to realsense-viewer
-- Check for updates capability added to realsense-depth-quality
-- Fetching recommended firmwares:
-- disable link time optimization for fw project
-- ... https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.15.0.2.bin
-- ... https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- ... https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
-- Configuring done
-- Generating done

-- Build files have been written to: /home/sha/software/librealsense/build

and I make install successfully，I still can not run realsense-viwer.

Carl-1999, 2023-06-30:
> Hi @Carl-1999 Have you solved this problem?

I haven't solved this problem yet.But I successfully installed L from the installation package. If you solve tthis problem ,please tell me.

MartyG-RealSense, 2023-06-30:
Are you able to find a built executable program for realsense-viewer if you go to the folder **usr/local/bin** where librealsense's examples and tools are located when they have been built from source code?

Carl-1999, 2023-07-01:
> Are you able to find a built executable program for realsense-viewer if you go to the folder **usr/local/bin** where librealsense's examples and tools are located when they have been built from source code?

Thank you for your reply！Yes.I can find it. When I go to the folder where I install , there is the output:
`sha@sha-TianHe:/usr/local/librealsense/bin$ ls
realsense-viewer   rs-distance           rs-pointcloud
rs-align           rs-embed              rs-post-processing
rs-align-advanced  rs-enumerate-devices  rs-record
rs-benchmark       rs-fw-logger          rs-record-playback
rs-callback        rs-fw-update          rs-rosbag-inspector
rs-capture         rs-gl                 rs-save-to-disk
rs-color           rs-hdr                rs-sensor-control
rs-convert         rs-hello-realsense    rs-software-device
rs-data-collect    rs-measure            rs-terminal
rs-depth           rs-motion
rs-depth-quality   rs-multicam

sha@sha-TianHe:/usr/local/librealsense/bin$ realsense-viewer
realsense-viewer：Command not found
`

MartyG-RealSense, 2023-07-01:
Does it run if you use the command **sudo realsense-viewer**

Carl-1999, 2023-07-02:
> Does it run if you use the command **sudo realsense-viewer**

It still doesn't work. I have already tried this command. Is it a kernel patch issue? 
When I build and apply patched kernel modules ,
run **./scripts/patch-realsense-ubuntu-lts-hwe.sh**,
 here will output **Reversed (or previously applied) patch detected! Assume -R? [n]** and I type **y**.
But this step will ultimately output success.
My kernel version is **5.15.0-75-generic**.
Another step that may cause errors may be **cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DBUILD_GRAPHICAL_EXAMPLES=TRUE -DCMAKE_INSTALL_PREFIX=/usr/local/librealsense**， here output ：
```
-- Fetching recommended firmwares:
-- disable link time optimization for fw project
-- ... https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.15.0.2.bin
-- ... https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- ... https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
-- Configuring done
-- Generating done
```
I don't know what this means,or Is it an issue with the SDK version?

MartyG-RealSense, 2023-07-02:
The messages `disable link time optimization for fw project` and `Reversed (or previously applied) patch detected!` do not seem to cause a problem during installation.  They have appeared in the installation logs of other RealSense users.

In regard to fetching recommended firmwares: librealsense recommends a firmware driver version to use with that specific SDK version for each type of RealSense camera and retrieves the recommended firmware files via https links.   So those messages are not something to be concerned about either.

It is rare to use the build flag `-DCMAKE_INSTALL_PREFIX` so you could try leaving that flag out of the CMake build instruction.

closed 2023-07-15 by MartyG-RealSense