realsenseai/librealsense #11928: Only rgb is not working in intel realsense d435i

Pradeep-Sudakar, 2023-06-20:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|233622075978|
|**Firmware Version**|05.14.00.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|226523131892|
|**Firmware Update Id**|226523131892|

Pradeep-Sudakar, 2023-06-20:
No @MartyG-RealSense it showing as D435i.

It doesnt work even after i only run RGB

[image]

MartyG-RealSense, 2023-06-20:
The SDK version listed at the top of the Viewer is **2.54.1** and your firmware version is listed as **5.14.0.0**.  This firmware version was designed for SDK 2.53.1.  The firmware that is recommended for 2.54.1 is **5.15.0.2**.

Please try downloading and running the 2.53.1 version of the Viewer from the 'Assets' file list at the link below using the **Intel.RealSense.Viewer.exe** link.  If RGB works in the 2.53.1 Viewer then this could indicate that you need to update your firmware to use SDK 2.54.1.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

[image]

Pradeep-Sudakar, 2023-06-20:
Im still getting no frames received:

[image]

MartyG-RealSense, 2023-06-20:
Please try resetting the camera to its faxtory-new default settings using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

If that does not work then next you can try uninstalling and reinstalling the camera drivers in the Windows Device Manager using the advice at https://github.com/IntelRealSense/librealsense/issues/10974#issuecomment-1270349796

Pradeep-Sudakar, 2023-06-20:
When i tried to reset it to its factory settings Im getting this error

[image]

Pradeep-Sudakar, 2023-06-20:
After uninstalling and reinstalling the drivers:

[image]

MartyG-RealSense, 2023-06-20:
Has the problem only recently developed or has the RGB not worked since the camera was purchased?

Pradeep-Sudakar, 2023-06-22:
The camera was initially working. But someone else was using it before me and i dont know the specs of the laptop that they were using.

MartyG-RealSense, 2023-06-22:
Are you using a laptop?  If so and it has a built-in RGB webcam in its screen then you could investigate the possibility that the webcam is blocking the RGB of the D435i.

You can test this by opening the Device Manager interface of Windows and expanding open the **Cameras** category, then finding the laptop camera (which may be called Integrated Camera).  Right-click on it and select 'Disable device' (not the Uninstall option).  This will temporarily disable the laptop webcam so you can see if the RealSense RGB now works.  If it does not make a difference then you can easily restore the laptop webcam by repeating the Device Manager process and selecting 'Enable device'.

[image]

closed 2023-07-05 by MartyG-RealSense