realsenseai/librealsense #11929: Get X,Y coordinates from X,Y,Z coordinates of realsense

BIGINCHI, 2023-06-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  -                                | 
| Platform                                 | PC  |
| SDK Version                            |  {  2 }                          | 
| Language                            |  {opencv/python}                          | 

### Issue Description
im using open3D to get x,y,z coordinates of the segments im detecting in my cloud . After detecting im converting the X,y,z coordinates to realsense coordinate systems , using the cloud points 

_def transform_point_from_realsense_open3d(cluster):
                    return([cluster[0][0],cluster[0][1], -cluster[0][2]])_

this snippet helps in converting open3d coordinates to realsense cloud coordinates 
now im using opencv_cloudpoint_viewer.py from the  wrappers/python/examples in librealsense 
i want to convert realsense X,Y,Z coordinates to X,Y coordinates to view on a picture that for which im using 

_cv2.imwrite('./out.png', color_image)_ in the opencv_cloudpoint_viewer.py

how do i get proper X,Y coordinates, from X,Y,Z coordinates and mark them using opencv circle function

MartyG-RealSense, 2023-06-20:
Hi @BIGINCHI  If the XYZ values of a pointcloud have been calculated with the SDK instruction **rs2_deproject_pixel_to_point** like line 72 of `opencv_pointcloud_viewer.py` does then the XYZ points can be converted back to XY pixels with the instruction **rs2_project_point_to_pixel**.  https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888 has a Python example of rs2_project_point_to_pixel.

In regard to marking with circles using OpenCV, the only case on tis forum that I could find is https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-613996151 where the RealSense user in that case uses OpenCV's **houghcircles** function.  Their script is at https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-614674313 and although it is a C++ script, the OpenCV sections of the code may provide some helpful ideas.

MartyG-RealSense, 2023-06-21:
Which RealSense camera model are you using, please?  

Most of the 400 Series models have a default Depth Unit Scale of **0.001**.  The D405 model has a default scale though of **0.01**.  

When RealSense data is used with a non-RealSense tool, the depth scale of the non-RealSense tool should be set to the same depth scale as the RealSense camera.  Otherwise, there will be incorrect scaling such as a pointcloud being too small.

BIGINCHI, 2023-06-21:
im using the D435i camera, sure ill try that. could you also share any references on how to play around with the scale and depth pointcloud of realsense. Thanks a Lot.

MartyG-RealSense, 2023-06-21:
The default depth unit scale of a RealSense camera can be changed to a custom value with the RealSense SDK instruction **rs.option.depth_units** as demonstrated in Python code at https://github.com/IntelRealSense/librealsense/issues/1905#issuecomment-398367795

BIGINCHI, 2023-06-22:
ok so one last doubt right now im saving images using the colour_image extracted from realsense bag file i want to view the 3D axes in that image what should i do?

BIGINCHI, 2023-06-22:
one last question, im saving the images of the realsense cloud point using

_cv2.imwrite('./out.png', color_image)_

[image]

what i want to include in the picture is an axes too which is within the cloud point

[image]

with this problem solved you can close the comment

MartyG-RealSense, 2023-06-22:
I researched your question but did not find a way to write axes onto an image when saving it, unfortunately.

BIGINCHI, 2023-06-22:
> I researched your question but did not find a way to write axes onto an image when saving it, unfortunately.

Thank you for the help. I have no further questions regarding this.

closed 2023-06-22 by MartyG-RealSense