realsenseai/librealsense #11932: Fail to connect after power reset

nathanieltagg, 2023-06-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.15 | 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  5.15                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.1 }                          | 
| Language                            |  C++ }                          | 
| Segment			|                        | 

### Issue Description
I cannot query the Realsense after doing a hardware reset.

```

      // Find the device.
      rs2::context context;
      rs2::device_list devices = context.query_devices();
      INFO << "Found " << devices.size() << " devices ";
      if(devices.size() == 0) {
        ERROR << "No realsense device found.";
        throw std::runtime_error("No realsense device found");
      }
     rs2::device device_ = devices[0];

     // Hardware reset
     if(hardware_reset_) {
        INFO << "Issuing hardware reset";
        device_.hardware_reset();        
        rs2::device_hub hub(context);
        device_ = hub.wait_for_device();
        INFO << "Hardware reset complete";
        usleep(1000000); // Does this help? No.
      }

      std::vector<rs2::sensor> sensors = device_.query_sensors();
```
I get the 'Hardware reset complete', and then the query_sensors call throws an error: "failed to set power state".

Is there any better method of doing the hardware reset? This was the only example I could find that didn't involve messing around with callbacks.

MartyG-RealSense, 2023-06-22:
Hi @nathanieltagg  An alternative to using the hardware reset instruction is to reset the entire USB port with an Ubuntu **bash** script.  This method of reset does not rely on the camera needing to be detected.  An example of such a script is at https://github.com/IntelRealSense/librealsense/issues/8393

MartyG-RealSense, 2023-06-28:
Hi @nathanieltagg  Do you require further assistance with this hardware reset case, please?  Thanks!