realsenseai/librealsense #11934: DllNotFoundException: realsense2 assembly MAC OS M1

CodinSegal, 2023-06-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | Latest| 
| Operating System & Version |   MacOS M1  | 
| Platform                                 | PC/Laptop |
| SDK Version                            |  SDK 2.0 (v2.54.1)                          | 
| Language                            |  C#/unity                    | 
| Segment			|  Robot                          | 

### Issue Description
Hello, So i am currently working on a project with robots of UR where we need to do a pick and place with object dection and we are using the intel sense d435, we are starting to integrate the part of intel sense in unity. So we downloaded the unity package and when I try to run it in unity 2021.3.2f on Mac M1 its says :

`DllNotFoundException: realsense2 assembly:<unknown assembly> type:<unknown type> member:(null)
Intel.RealSense.Context..cctor () (at <174f4594684e480d948f375945ed1b21>:0)
Rethrow as TypeInitializationException: The type initializer for 'Intel.RealSense.Context' threw an exception.
RsDeviceListener+<Start>d__10.MoveNext () (at Assets/RealSenseSDK2.0/Misc/Utils/RsDeviceListener.cs:27)
UnityEngine.SetupCoroutine.InvokeMoveNext (System.Collections.IEnumerator enumerator, System.IntPtr returnValueAddress) (at /Users/bokken/buildslave/unity/build/Runtime/Export/Scripting/Coroutines.cs:17)`

Do you guys have any suggestion to get this part up and running ?

MartyG-RealSense, 2023-06-23:
Hi @CodinSegal  In the detailed guide for installation of RealSense on Mac (including M1 models) at the link below, the comments section at the bottom of the page has advice on getting it working with Unity.

https://lightbuzz.com/realsense-macos/

CodinSegal, 2023-06-26:
Hello @MartyG-RealSense, thank you for the suggestion. I followed this tutorial before encountering the issue, but unfortunately, I still can't get it to work on my MacBook Pro. It seems like it might not be compatible with macOS. If you have any other resources or suggestions, I would greatly appreciate it.

MartyG-RealSense, 2023-06-26:
That tutorial is the only guide available for using Unity with modern MacOS versions, unfortunately.  

RealSense Unity wrapper installation guidance for Mac is also shared at https://github.com/IntelRealSense/librealsense/issues/1576#issuecomment-390395111 although it is 5 years old and for Intel based Macs rather than the Apple Silicon M1 models of Mac.

There are also currently camera detection problems with the recent MacOS Ventura at the time of writing this.  If you are using Ventura then https://github.com/IntelRealSense/librealsense/issues/11815 is the central place on this support forum where I am directing Ventura users who are having problems so that they can discuss it amongst themselves, and you will therefore be notified automatically of any solutions that the RealSense Mac community comes up with for getting librealsense working on Ventura if you have left a comment there.

MartyG-RealSense, 2023-07-03:
Hi @CodinSegal  Do you have an update about this case that you can provide, please?  Thanks!

CodinSegal, 2023-07-03:
@MartyG-RealSense, thanks for all the links, I have tried to fallow all the references you gave me and i still had problem setting up even out side of unity. So I had to move into a linux enviorment, where I have been using it in a python evironment. Thank you for your support.

MartyG-RealSense, 2023-07-03:
You are very welcome.  Thanks very much for the update.  As you moved from Mac to Linux, I will close this case.  Thanks again!

closed 2023-07-03 by MartyG-RealSense