realsenseai/librealsense #11937: d455 in ubuntu. Alignment of images acquired by infrared and color cameras, color image acquisition by infrared cameras, rough acceleration of gravity

KOBOSP, 2023-06-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.15.0-75-generic   | 
| Platform                                 | i7-12700 |
| SDK Version                            |  2.54.1                         | 
| Language                            |  labview                          | 
| Segment			|  {others }                          | 

### Issue Description
Question 1: When the viewer is set as shown in Figure 1, the rqt_bag interface in Figure 2 shows that the color image and the infrared image are not completely aligned, and there is a time interval of about 0.0585s between them (this time interval is a random value, it are different across multiple records). May I ask whether the interval can be completely aligned, and if so, how should it be set in the viewer.
[image]
[image]

Question 2: I want to get the color image from the left IR camera and image in any format (infrared or color) from the right IR camera at the same time. However, as shown in Figure 3, such a setting in the Viewer will cause errors. Can this configuration be realized?
[image]
[image]

Question 3: According to the table shared on the Internet, the typical acceleration of gravity at the location of the device is 9.7947m/s2. But Figure 4 shows that the gravitational acceleration measured by the IMU in a static state is 9.62-9.63. Is this normal? If not, how to calibrate it with low-cost measures.
[image]

MartyG-RealSense, 2023-06-25:
Hi @KOBOSP  The RGB and Infrared images are not aligned because the different types of sensors on the front of the camera are in different physical positions, horizontally offset from each other.  The RGB sensor is next to the left infrared sensor and so while they have a similar field of view, they are not viewing the scene from exactly the same position because of this offset.

The RealSense Viewer does not provide an alignment function in its **2D** mode (shown in your images above).  Its **3D** pointcloud mode can map together depth and RGB or depth and infrared, but not RGB and infrared.

RGB8 color is only supported from the left infrared sensor and the right infrared sensor only provides Y8 infrared, so it is not possible unfortunately to stream RGB8 color from both the left and right infrared sensors as doing so will cause the pop-up error that you hghlighted.  You can stream infrared RGB and the RGB sensor at the same time, but they will not have exactly the same viewpoint, and the RGB from left infrared also looks different in its coloring to the RGB stream.

At https://github.com/IntelRealSense/librealsense/issues/1548#issuecomment-389070550 a RealSense team member advises that when two separate sensors are being streamed, they are not perfectly aligned but the RealSense SDK tries to find the best match between the timings.  I note that you have the RGB option **Auto Exposure Priority** disabled, so the SDK should be trying to enforce a constant FPS speed for all streams.

KOBOSP, 2023-06-26:
Supplement to question 2: Can the left infrared sensor output RGB8 and the right infrared sensor output Y8 simultaneously on the d455?

MartyG-RealSense, 2023-06-26:
No, if the left sensor is streaming RGB8 color then the right sensor cannot stream Y8 at the same time.

KOBOSP, 2023-06-26:
thanks for your clarification

closed 2023-06-26 by KOBOSP