realsenseai/librealsense #11938: Another Undefined symbols for architecture x86_64 on MacOS

caddac, 2023-06-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   MacOS  | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/CPP                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm basically having the same issue as https://github.com/IntelRealSense/librealsense/issues/2685 (though for a different dependency). 

I have the `rs-hello-realsense.cpp` example in a directory (`~/src/realsense/rs-hello-realsense.cpp`).
I run `cmake`, then from the build directory `make` and get the `Undefined symbols for architecture x86_64` error.

```
╭─ ~/src/realsense                                                                                                                                                                    ✔  9099  16:06:39
╰─ cmake -S . -B build
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/dave/src/realsense/build
╭─ ~/src/realsense                                                                                                                                                                    ✔  9100  16:06:42
╰─ cd build
╭─ ~/src/realsense/build                                                                                                                                                              ✔  9101  16:06:46
╰─ make
[ 50%] Linking CXX executable rs-hello-realsense
Undefined symbols for architecture x86_64:
  "_rs2_create_context", referenced from:
      rs2::context::context() in rs-hello-realsense.cpp.o
  "_rs2_create_pipeline", referenced from:
      rs2::pipeline::pipeline(rs2::context) in rs-hello-realsense.cpp.o
  "_rs2_delete_context", referenced from:
      rs2::context::context() in rs-hello-realsense.cpp.o
  "_rs2_delete_pipeline", referenced from:
      rs2::pipeline::pipeline(rs2::context) in rs-hello-realsense.cpp.o
  "_rs2_delete_pipeline_profile", referenced from:
      rs2::pipeline::start() in rs-hello-realsense.cpp.o
  "_rs2_depth_frame_get_distance", referenced from:
      rs2::depth_frame::get_distance(int, int) const in rs-hello-realsense.cpp.o
  "_rs2_embedded_frames_count", referenced from:
      rs2::frameset::frameset(rs2::frame const&) in rs-hello-realsense.cpp.o
  "_rs2_extract_frame", referenced from:
      void rs2::frameset::foreach_rs<rs2::frameset::first_or_default(rs2_stream, rs2_format) const::'lambda'(rs2::frame)>(rs2::frameset::first_or_default(rs2_stream, rs2_format) const::'lambda'(rs2::frame)) const in rs-hello-realsense.cpp.o
  "_rs2_frame_add_ref", referenced from:
      rs2::frame::add_ref() const in rs-hello-realsense.cpp.o
  "_rs2_free_error", referenced from:
      rs2::error::error(rs2_error*) in rs-hello-realsense.cpp.o
  "_rs2_get_error_message", referenced from:
      rs2::error::error(rs2_error*) in rs-hello-realsense.cpp.o
  "_rs2_get_failed_args", referenced from:
      rs2::error::error(rs2_error*) in rs-hello-realsense.cpp.o
  "_rs2_get_failed_function", referenced from:
      rs2::error::error(rs2_error*) in rs-hello-realsense.cpp.o
  "_rs2_get_frame_height", referenced from:
      rs2::video_frame::get_height() const in rs-hello-realsense.cpp.o
  "_rs2_get_frame_stream_profile", referenced from:
      rs2::frame::get_profile() const in rs-hello-realsense.cpp.o
  "_rs2_get_frame_width", referenced from:
      rs2::video_frame::get_width() const in rs-hello-realsense.cpp.o
  "_rs2_get_librealsense_exception_type", referenced from:
      rs2::error::handle(rs2_error*) in rs-hello-realsense.cpp.o
      rs2::error::error(rs2_error*) in rs-hello-realsense.cpp.o
  "_rs2_get_stream_profile_data", referenced from:
      rs2::stream_profile::stream_profile(rs2_stream_profile const*) in rs-hello-realsense.cpp.o
  "_rs2_is_frame_extendable_to", referenced from:
      rs2::frameset::frameset(rs2::frame const&) in rs-hello-realsense.cpp.o
      rs2::depth_frame::depth_frame(rs2::frame const&) in rs-hello-realsense.cpp.o
      rs2::video_frame::video_frame(rs2::frame const&) in rs-hello-realsense.cpp.o
  "_rs2_is_stream_profile_default", referenced from:
      rs2::stream_profile::stream_profile(rs2_stream_profile const*) in rs-hello-realsense.cpp.o
  "_rs2_pipeline_start", referenced from:
      rs2::pipeline::start() in rs-hello-realsense.cpp.o
  "_rs2_pipeline_wait_for_frames", referenced from:
      rs2::pipeline::wait_for_frames(unsigned int) const in rs-hello-realsense.cpp.o
  "_rs2_release_frame", referenced from:
      rs2::frame::reset() in rs-hello-realsense.cpp.o
      rs2::frame::~frame() in rs-hello-realsense.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [rs-hello-realsense] Error 1
make[1]: *** [CMakeFiles/rs-hello-realsense.dir/all] Error 2
make: *** [all] Error 2
```

From #2685 it looks like I'm maybe missing linking some libraries, but the `g++ ...` command suggested doesn't work:
```
g++ -I /Users/dave/src/librealsense/include/ -I ../ -std=c++11 ../rs-hello-realsense.cpp -o rs-hello-realsense -lglfw -lrealsense2 -framework OpenGL
ld: library not found for -lglfw
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Suggestions for how to resolve this?

caddac, 2023-06-26:
discovered that `glfw` didn't exist in `/usr/local/lib`, so installed it using `brew install glfw` and now I"m getting:

```
╭─ ~/src/realsense/build                                                                                                                                                            2 ↵  9124  19:25:15
╰─ g++ -I /Users/dave/src/librealsense/include/ -I ../ -std=c++11 ../rs-hello-realsense.cpp -o rs-hello-realsense -lglfw -lrealsense2 -framework OpenGL
ld: library not found for -lrealsense2
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

caddac, 2023-06-26:
Realized I missed the last step, maybe? Ran the `install_name_tool` command from the "Packaging your Application" [step](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md#packaging-your-application), but I'm unclear about where to put the resulting dylib file.

MartyG-RealSense, 2023-06-26:
Hi @caddac  Do you have an M1 model of Mac, please?  If you do then the M1 processor chip is based on Arm architecture, whilst older Macs with Intel-based processors are based on x86/x64 architecture.  The error message **Undefined symbols for architecture x86_64** suggests to me that your Mac is not an old Intel-based model.

The link below has a guide for both Intel and M1 Macs about building the librealsense SDK.

https://lightbuzz.com/realsense-macos/

The SDK can alternatively be installed for Intel and M1 Macs and MacOS versions up to Ventura with **brew install librealsense**

https://formulae.brew.sh/formula/librealsense

caddac, 2023-06-26:
Hi @MartyG-RealSense Sorry for not specifying, I am running Monterey on an Intel Mac.
I'll take a look at the lightbuzz article and report back! 
Thank you for the suggestions!

caddac, 2023-06-30:
I ran into needing to upgrade Xcode, but my HDD is full. Its going to take me a while to get a new HDD and update Xcode. Closing this for now and will re-open when I get back to this.

closed 2023-06-30 by caddac

MartyG-RealSense, 2023-07-01:
That's no problem at all to re-open later.  Thanks very much for the update.  Good luck!