realsenseai/librealsense #11939: Dynamic Calibratior underwater

hfdlwt, 2023-06-26:
Hi ！ I have a question, which material is required to print the DynamicCalibrator target when using DynamicCalibrator underwater? What should I pay attention to when printing? How can Dynamic Calibrator be used underwater to complete calibration faster?
[https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/print-target-fixed-width.pdf](url)

MartyG-RealSense, 2023-06-26:
Hi @hfdlwt  A pre-made Dynamic Calibration target board product made from special material is available from Intel for $1500 USD that is suitable for use in large underwater spaces such as pools.  Because of this special material (information is not available about the material used) this large-scale target cannot be self-printed on paper by RealSense users.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html
 
For smaller underwater spaces such as fish aquariums, the free downloadable print target should be sufficient.

https://dev.intelrealsense.com/docs/dynamic-calibration-print-target

If you are printing the target yourself for underwater use then it should presumably be enclosed in a rigid plastic see-through sleeve so that the surface remains flat like when mounted on a wall.  Googling for the term **rigid document holder waterproof** will provide some examples of possible products to seal the printed target in.

There is an extensive discussion about underwater calibration with the Dynamic Calibration tool at https://github.com/IntelRealSense/realsense-ros/issues/1723

The discussion is so long that a lot of it is hidden under a cut, so I recommend going to https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-802741234 first and clicking the '25 hidden items' link to show the entire discussion.

Once the full discussion is uncovered, to shorten reading time I also recommend reading the advice by RealSense team member **agrunnet** at https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-808406783 and the comments beneath it.

hfdlwt, 2023-06-28:
Thank you @MartyG-RealSense , but I have another question. Should I need to print the gray area in the middle? If there are no gray areas, is there any impact?

MartyG-RealSense, 2023-06-28:
Yes, in order for the Dynamic Calibration to work you should print the entire page 1 of the PDF target document without editing it, keeping everything including the text on the page.  Page 2 does not need to be printed.

hfdlwt, 2023-06-28:
Thanks @MartyG-RealSense  , Can I customize the calibration board with materials such as aluminum oxide?

MartyG-RealSense, 2023-06-28:
I am not aware of othe RealSense users who have attempted this but as long as the materials do not make the target harder for the camera to read (for example, by being reflective or colored black) then I do not see a reason why it would not be possible.

closed 2023-07-11 by MartyG-RealSense