realsenseai/librealsense #11942: realsens d435 ExternalException: rs2_pipeline_wait_for_frames(pipe:000001F888C12C70) Rethrow as Exception: Frame didn't arrive within 5000

ryoya-tamura, 2023-06-27:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Win (11)   | 
| Platform                                 | PC |
| SDK Version                            |   2.0                           | 
| Language                            |  C#//unity                           | 
| Segment			|  VR/AR                          | 

### Issue Description
Sorry if this may be unnatural as I am not an English speaker at the beginning.

I am working on a function that uses the D435 to display a video clip of an object on the screen only when the object comes within 1 meter.
However, an error occurs.
ExternalException: rs2_pipeline_wait_for_frames(pipe:000001F888C12C70)
Rethrow as Exception: Frame didn't arrive within 5000

Here is the code.
```
using UnityEngine;

public class NotionDepth : RsDevice
{
    [SerializeField] GameObject canvas; 
    void Update()
    {

        if (!Streaming)
            return;

        if (processMode != ProcessMode.UnityThread)
            return;

        if (m_pipeline.PollForFrames(out var frames))
        {
        //Debug.Log(frames.DepthFrame.GetDistance(320, 240));
        
        double p = frames.DepthFrame.GetDistance(320, 240);

        Debug.Log(p);

        //[D]キーを押す
        if (1.000 < p) //Input.GetKeyDown(KeyCode.D)
        {
            //ゲームオブジェクト表示→非表示
            canvas.SetActive(false);
            Debug.Log("1m以上外");
        }
        else //Input.GetKeyDown(KeyCode.A)
        {
            canvas.SetActive(true);
            Debug.Log("1m以内");
        }

        }
    }
        protected override void WaitForFrames()
    {
        while (!stopEvent.WaitOne(0))
        {
            using (var frames = m_pipeline.WaitForFrames())

                //[D]キーを押す
                if (1.000 < frames.DepthFrame.GetDistance(320, 240)) //Input.GetKeyDown(KeyCode.D)
                {
                //ゲームオブジェクト表示→非表示
                //canvas.SetActive(false);
                Debug.Log("1m以上外");
                }
                else //Input.GetKeyDown(KeyCode.A)
                {
                //canvas.SetActive(true);
                Debug.Log("1m以内");
                }

        }
    }
}
```
It is based on BackgroundSegmentation of SDK.
See the image for the project hierarchy and code attachments.
[image]

Please let me know if I am missing some necessary information.
Thank you!

MartyG-RealSense, 2023-06-27:
Hi @ryoya-tamura  Did you set up your Unity project by importing the **Intel.RealSense.unitypackage** file?  If you did, which version of the RealSense SDK is your UnityPackage file from, please?  Is it 2.53.1 or 2.54.1.

[image]

If you are using the file from 2.53.1 then firmware driver **5.14.0.0** should be used.  If it is 2.54.1 then the 5.15.0.2 firmware that you currently have installed in the camera is the correct one.

ryoya-tamura, 2023-06-27:
Thank you very much for your reply.
I imported the Intel.RealSense.unitypackage file and set up a Unity project.
I am using a package installed on 6/16, so I believe it is the latest 2.54.1.
Please let me know if there is a way to check the version from unity.

MartyG-RealSense, 2023-06-27:
It is not possible to check the SDK version from within the Unity wrapper.

If you downloaded the file on the 16th June then it is probable that it was obtained from the latest 2.54.1 version.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

ryoya-tamura, 2023-06-28:
I tried re-installing version 2.54.1, but the result was the same.

I did not tell them that I had made some changes to RsDevice.cs. Sorry.

Line 63.
`protected readonly AutoResetEvent stopEvent = new AutoResetEvent(false);`
Line 64.
`protected Pipeline m_pipeline;.`
Line 155.
`protected virtual void WaitForFrames()`

MartyG-RealSense, 2023-06-28:
As the original lines were **private**, what happens if you change them to **public** instead of protected?

ryoya-tamura, 2023-06-29:
Thank you very much for your reply.
I modified the code as follows, but the result remains the same.

` public override void WaitForFrames()`

`RsDevice.cs`
Line 63.
`public readonly AutoResetEvent stopEvent = new AutoResetEvent(false);`
Line 64.
`public Pipeline m_pipeline;.`
Line 155.
`public virtual void WaitForFrames()`

ryoya-tamura, 2023-06-29:
ReDevice's Process Mode is set to Unity Thread. As a result, the error disappeared.

However, the Depth Segmentation video is not displayed in real time.
The problem is not solved because I want to use depth information to decide whether to display real-time video or not.

Thank you.

https://github.com/IntelRealSense/librealsense/assets/134613854/ec7dd5fe-7a1f-4159-a01a-2dd15ce1d0bf

MartyG-RealSense, 2023-06-29:
I tried the BackgroundSegmentation example in Unity Thread mode in Unity 2021 and it was able to display normally.  I note though that your project was **based** on BackgroundSegmentation.  Is the original BackgroundSegmentation scene able to run in Unity Thread mode on your computer?

ryoya-tamura, 2023-06-29:
Thank you very much for your reply.
Yes. The original BackgroundSegmentation scene is able to run in Unity Thread mode on my computer.

MartyG-RealSense, 2023-06-29:
In https://github.com/IntelRealSense/librealsense/issues/11942#issuecomment-1610810229 you modified RsDevice.cs.  Since this file will be used by all RealSense examples and not just BackgroundSegmentation, it could be concluded that if BackgroundSegmentation still works after the RsDevice changes then the changes are not responsible for your **Frame didn't arrive within 5000** problem.

MartyG-RealSense, 2023-07-05:
Hi @ryoya-tamura  Do you require further assistance with this case, please?  Thanks!

ryoya-tamura, 2023-07-06:
The problem has not been solved, but we will try to achieve our goals in other ways. Thanks.

closed 2023-07-06 by ryoya-tamura