realsenseai/librealsense #11945: Compatible librealsense package with realsense-ros for ros2-hkr?

kenloumixx, 2023-06-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am trying to build [realsense-ros2 of version ros2-hkr](https://github.com/IntelRealSense/realsense-ros/tree/ros2-hkr) with librealsense 2.54.1 installed. When `colcon build`, realsense2_camera package fails to be built with the error 
   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
I'd like to figure out how to find the matched version of librealsense. 
TIA

MartyG-RealSense, 2023-06-28:
Hi @kenloumixx  As that ros2-hkr branch's **realsense2_camera** is dated at 2 weeks old at the time of writing this, it is likely intended for use with librealsense **2.51.1** rather than 2.54.1, as the latest ROS wrapper **4.51.1** was only released yesterday.

https://github.com/IntelRealSense/realsense-ros

kenloumixx, 2023-06-28:
Thank you for the answer! If I understand correctly, you mean that I should use librealsense 2.51.1 and Ros wrapper 4.54.1?? I installed the librealsense following the manual in [this link](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md), but librealsense 2.54.1 is installed. How can I specify the version of librealsense?

MartyG-RealSense, 2023-06-28:
The matching ROS wrapper for 2.51.1 is wrapper 4.51.1.  

Usually the librealsense version to match to is listed in the 'Supported RealSense SDK' section of the release notes for a particular wrapper version.  The notes for wrapper 4.51.1 can be found at the link below.

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

[image]

Information about specifying a particular librealsense version to install instead of the latest one can be found at https://github.com/IntelRealSense/librealsense/issues/10526

kenloumixx, 2023-06-30:
Thank you for the response :) Well I run the command `ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true pointcloud.ordered_pc:=true pointcloud.enable:=true initial_reset:=true` with the right versions of librealsense and realsense-ros wrapper, but it shows the warning 
```
[realsense2_camera_node-1] [WARN] [1688128656.710690918] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1688128656.890759458] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1688128657.629850832] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1688128658.100398240] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1688128659.462177329] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1688128659.979261125] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1688128662.467378269] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1688128663.858762725] [camera.camera]: No stream match for pointcloud chosen texture Process - Color

``` 
with no messages on the topic `/camera/depth/color/points`..

kenloumixx, 2023-06-30:
I am using Depth Camera 435.

MartyG-RealSense, 2023-06-30:
Thanks for the model confirmation.  So color frames should be available on D435 as it has an RGB sensor.

Yes, color frames are RGB frames.

Does the` no stream match` warning still occur if a low color and depth resolution of 648x480 for both is defined?

ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true pointcloud.ordered_pc:=true pointcloud.enable:=true initial_reset:=true depth_module.profile:=640x480x30 rgb_camera.profile:=640x480x30

kenloumixx, 2023-07-01:
`No stream match for pointcloud chosen texture Process - Color` has gone but other messages including resolution related error appear as below: 

```
[realsense2_camera_node-1] [WARN] [1688190474.401992098] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [ERROR] [1688190474.467793041] [camera.camera]: Given value, 640x480x3 is invalid. Set ROS param back to: 1280x720x30
[realsense2_camera_node-1] [INFO] [1688190474.472356258] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1688190474.473134731] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1688190474.525570920] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1688190474.580219324] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1688190474.640004393] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1688190474.641166998] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1688190474.652182510] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1688190474.661481396] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1688190474.663226471] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1688190474.709018939] [camera.camera]: 
[realsense2_camera_node-1]  01/07 05:48:31,591 ERROR [547205671232] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-1]  01/07 05:48:31,591 ERROR [547205671232] (error-handling.cpp:94) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-1]  01/07 05:48:31,632 ERROR [547151620416] (backend-v4l2.cpp:330) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 18 error: No such device
[realsense2_camera_node-1]  01/07 05:48:32,371 ERROR [547180493120] (librealsense-exception.h:52) acquire: Cannot open '/dev/video0 Last Error: No such file or directory
[realsense2_camera_node-1]  01/07 05:48:32,372 ERROR [547180493120] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video0 Last Error: No such file or directory
[realsense2_camera_node-1]  01/07 05:48:32,592 ERROR [547205671232] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-1]  01/07 05:48:32,592 ERROR [547205671232] (error-handling.cpp:94) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-1]  01/07 05:48:33,372 ERROR [547180493120] (librealsense-exception.h:52) acquire: Cannot open '/dev/video0 Last Error: No such file or directory
[realsense2_camera_node-1]  01/07 05:48:33,372 ERROR [547180493120] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video0 Last Error: No such file or directory
[realsense2_camera_node-1]  01/07 05:48:33,595 ERROR [547205671232] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-1]  01/07 05:48:33,595 ERROR [547205671232] (error-handling.cpp:94) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[realsense2_camera_node-1]  01/07 05:48:34,373 ERROR [547180493120] (librealsense-exception.h:52) acquire: Cannot open '/dev/video0 Last Error: No such file or directory
[realsense2_camera_node-1]  01/07 05:48:34,373 ERROR [547180493120] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video0 Last Error: No such file or directory
```

MartyG-RealSense, 2023-07-01:
A RealSense ROS user created a RealSense Docker for the RealSense ROS2 wrapper **ros2-development** at the link below that is only 4 months old at the time of writing this, so it may be a helpful reference.

https://github.com/2b-t/realsense-ros2-docker

kenloumixx, 2023-07-04:
@MartyG-RealSense Thank you for response :) I'll take a good look the link you shared and communicate with the author! Thank you :)

closed 2023-07-04 by kenloumixx