realsenseai/librealsense #11948: Bug in python SDK. Not able to connect when the camera USB connected by extension cable

dsnsabari, 2023-06-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):
 

* All users are welcome to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D455 } | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Windows10  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / (2.54.1.5217) }                          | 
| Language                            |  {python/unity }                          | 
| Segment			|  {Real time object distance analysis }                          | 

### Issue Description

I am getting the below error when I connect the camera USB to the [Extension Cable](https://www.amazon.co.jp/dp/B01LA56WDM?ref_=cm_sw_r_cp_ud_dp_AW7B7FB4B8K81541QVA3&th=1)

I was trying the get the RGB and depth image using the D455. I will use the depth image for object distance measurement. I can see the RGB and depth image in Intel Realsense when I connect the camera USB to the Extension Cable. But when I try to get the RGB image/depth image from Python, it throws the below error.  Please help me in fixing the issue. 

```
Exception has occurred: RuntimeError
Couldn't resolve requests
  File "D:\Depth_Camera\realtime_testing.py", line 48, in <module>
    pipeline_profiles[device] = pipeline.start(config)

```

**Python code used for testing:**

```
import pyrealsense2 as rs
import numpy as np
import cv2

connect_device = []

for d in rs.context().devices:
    if d.get_info(rs.camera_info.name).lower() != 'platform camera':
        serial = d.get_info(rs.camera_info.serial_number)
        product_line = d.get_info(rs.camera_info.product_line)
        device_info = (serial, product_line)  # (serial_number, product_line)
        connect_device.append(serial)
        
# Define the criteria for corner detection
#criteria = (cv2.TERM_CRITERIA_EPS + cv2.TERM_CRITERIA_MAX_ITER, 30, 0.001)
#align = rs.align(rs.stream.color)

pipelines = {}
pipeline_profiles = {}
for device in connect_device:
    config = rs.config()
    config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)
    # config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 5)
    # config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 5)
    # Align objects
    align_to = rs.stream.color
    align = rs.align(align_to)
    pipeline = rs.pipeline()
    config.enable_device(device)

    pipeline_profiles[device] = pipeline.start(config)
    depth_scale = pipeline_profiles[device].get_device().first_depth_sensor().get_depth_scale()
    pipelines[device] = pipeline

```

**Intel RealSense Viewer Output:**

I cropped person details from the RGB.  

[image]

MartyG-RealSense, 2023-06-28:
Hi @dsnsabari  Because you are using a USB 2 cable, the camera will operate in USB 2 mode even if the cable is plugged into a USB 3 port.  The supported resolution and FPS combinations that are supported on USB 2 are more limited.  For example, 640x480 at 5 FPS (or up to 30 FPS) is supported but the 1280x720 resolution will only support 5 FPS on a USB 2 connection on D455.  USB 3 is required for 30 FPS at that resolution.

If a requested configuration cannot currently be provided by the camera then the **Couldn't resolve requests** error is generated.

dsnsabari, 2023-06-29:
Hi @MartyG-RealSense, It is working now.

closed 2023-06-29 by dsnsabari