realsenseai/librealsense #11950: Segmentation fault when there are other GMSL cameras connected besides the D457

dmellosanjay, 2023-06-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | 5.15.00.02 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Jetson Orin  |
| SDK Version                            |  2.54.1                         | 
| Language                            | C++                        | 

### Issue Description

I am unable to run any of the example applications to test the D457 when there is another GMSL camera connected.
```
./rs-hello-realsense 
Segmentation fault (core dumped)
```

Here's the output of `v4l2-ctl` showing 2 camera modules (D457 and an ar0233)connected to the Jetson
```
$ v4l2-ctl --list-devices
NVIDIA Tegra Video Input Device (platform:tegra-camrtc-ca):
	/dev/media0

vi-output, DS5 mux 72-001b (platform:tegra-capture-vi:0):
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6

vi-output, ar0233 42-0044 (platform:tegra-capture-vi:2):
	/dev/video0
```

I verified that I am able to stream frames from the depth camera using `v4l2-ctl`
```
$ v4l2-ctl --stream-count=100 --stream-mmap --stream-to=somefile.raw -d /dev/video1
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.00 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.00 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.00 fps
```

Also, verified that the example applications run without issues with only the D457 connected

```
./rs-hello-realsense 
The camera is facing an object 2.933 meters away
```

MartyG-RealSense, 2023-06-29:
Hi @dmellosanjay  I will discuss your issue with my Intel RealSense colleaues.  Thanks very much for your patience.

hsuys, 2023-07-06:
Hi @dmellosanjay , if possible, can you share the output of `rs-enumerate-devices -S` when two cameras are connected?

dmellosanjay, 2023-07-06:
Hi @hsuys it segfaults. 

```root@orin:~/# rs-enumerate-devices 
Segmentation fault (core dumped)

root@orin:~/# v4l2-ctl --list-devices
NVIDIA Tegra Video Input Device (platform:tegra-camrtc-ca):
	/dev/media0

vi-output, DS5 mux 68-001a (platform:tegra-capture-vi:0):
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6

vi-output, ar0233 42-0044 (platform:tegra-capture-vi:2):
	/dev/video0
```

hsuys, 2023-07-06:
Thank you. 

> vi-output, ar0233 42-0044 (platform:tegra-capture-vi:2):
> 	/dev/video0

Perhaps this is the issue. Please see:
https://github.com/IntelRealSense/librealsense/blob/master/src/linux/backend-v4l2.cpp#L757

Can you please try and discard the non-realsense devices by checking the device name? Something like:
```
if (card.find("DS5 mux") != std::string::npos || card.find("vi-metadata") != std::string::npos )
{
	//std::cout <<"Realsense device ";
}
else
{
    LOG_WARNING("Unresolved RealSense device pattern: " << name << ", device is skipped");
     throw linux_backend_exception("Unresolved RealSense device, device is skipped");
} 
```

dmellosanjay, 2023-07-10:
@hsuys Thanks, that fixed it!

Hope this is added in the next release.

dmellosanjay, 2023-07-10:
This is the output of `rs-enumerate-devices` after adding the check for non-D457 devices
```
# rs-enumerate-devices -S
 10/07 22:44:42,545 ERROR [281473799768320] (librealsense-exception.h:52) Unresolved RealSense device, device is skipped Last Error: No such file or directory
 10/07 22:44:42,553 ERROR [281473799768320] (librealsense-exception.h:52) Unresolved RealSense device, device is skipped Last Error: No such file or directory
 10/07 22:44:42,562 ERROR [281473799768320] (librealsense-exception.h:52) Unresolved RealSense device, device is skipped Last Error: No such file or directory
 10/07 22:44:42,567 ERROR [281473799768320] (librealsense-exception.h:52) Unresolved RealSense device, device is skipped Last Error: No such file or directory
 10/07 22:44:42,572 ERROR [281473799768320] (librealsense-exception.h:52) Unresolved RealSense device, device is skipped Last Error: No such file or directory
 10/07 22:44:42,736 ERROR [281473799768320] (librealsense-exception.h:52) Unresolved RealSense device, device is skipped Last Error: No such file or directory
Segmentation fault (core dumped)
```

hsuys, 2023-07-13:
@dmellosanjay , thanks and I hope the fix provides temporary fix and unblock your current development. The segfault on rs-enumerate-devices will need to be further looked into.

MartyG-RealSense, 2023-07-20:
Hi @dmellosanjay  As Intel will look into the rs-enumerate-devices segfault that you reported, do you require further assistance with this case please?  Thanks!

dmellosanjay, 2023-07-21:
@MartyG-RealSense No further assistance needed for this, thank you.

MartyG-RealSense, 2023-07-21:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-07-21 by MartyG-RealSense