realsenseai/librealsense #11951: HDR not supported on D457

dmellosanjay, 2023-06-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | 5.15.00.02 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.10.104 / Jetpack 5.0.2                                       | 
| Platform                                 | NVIDIA Jetson Orin |
| SDK Version                            | 2.54.1                       | 
| Language                            |  C++                       | 

### Issue Description

Version 2.54.1 of the SDK does not support any HDR for the D457.  Is this expected to be supported in the next release? 

Output from the HDR example
```
./rs-hdr 
Firmware and/or SDK versions must be updated for the HDR feature to be supported.
```

Verified that the camera is on the latest firmware
```
# rs-enumerate-devices -S
Device Name                   Serial Number       Firmware Version    
Intel RealSense D457          220422301683        5.15.0.2            
Device info: 
    Name                          : 	Intel RealSense D457
    Serial Number                 : 	220422301683
    Firmware Version              : 	5.15.0.2
    Recommended Firmware Version  : 	5.15.0.2
```

MartyG-RealSense, 2023-06-29:
Hi @dmellosanjay  I believe that you are referring to the example below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/hdr

I tested HDR with D457 in the 2.54.1 version of the RealSense Viewer with firmware 5.15.0.2 and was able to enable HDR without problems.  This was on a USB connection though as I do not have a GMSL setup.

The response of rs-hdr about updating SDK and firmware does not sound correct, as the HDR white-paper document states that the minimum requirements for the HDR feature are firmware 5.12.8.100 or newer and SDK 2.39.0 or newer.

You could try editing out the 'else' check in rs-hdr that generates this message and stops the program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/hdr/rs-hdr.cpp#L51C4-L55

dmellosanjay, 2023-07-03:
@MartyG-RealSense The D457-specific checks aren't just in this file though. Any HDR-related configuration on the D457 is not supported (even disabling HDR is not allowed). Also, the camera enumerates as a D455 when it's in USB mode which explains why it works in USB but not GMSL.

Anyway, here's the output after commenting out the else block in the file you mentioned.
```
# ./rs-hdr 
RealSense error calling rs2_set_option(options:0xaaaad8719a20, option:Sequence Name, value:0):
    object doesn't support option #77
```

MartyG-RealSense, 2023-07-06:
I have been informed that the HDR function (rs-hdr) is not supported when the D457 is in MIPI mode instead of the D455 USB mode, as indicated in the code highlighted in the SDK's **test-hdr.cpp** file at the link below.
 
https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/live/d400/test-hdr.cpp#L8

dmellosanjay, 2023-07-06:
@MartyG-RealSense Thanks. Will it be supported on the D457 in a release soon? This is an important feature for our use-case.

MartyG-RealSense, 2023-08-04:
Hi @dmellosanjay  My understanding is that one of my Intel RealSense colleagues followed up with you about the question of HDR support for D457.  If they did and you are satisfied, can this issue now be closed?  Thanks!

dmellosanjay, 2023-08-04:
@MartyG-RealSense Yes, it can be closed, thanks.

MartyG-RealSense, 2023-08-04:
Great, thanks very much for the update!

closed 2023-08-04 by MartyG-RealSense

MartyG-RealSense, 2023-09-28:
Hi again @dmellosanjay   In the latest 2.54.2 release of the SDK, rs-hdr has been bugfixed to include support for D457.  Support for using rs-fw-update to update D457 firmware is also in this release.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2542