realsenseai/librealsense #11954: Documentation change: repo key location

Comnenus, 2023-06-29:
Following the instructions here: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

I came across a problem where the repo is not signed despite the pubkey having been downloaded.  I was able to fix this with the following changes:

```
$ curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | sudo tee /etc/apt/trusted.gpg.d/librealsense.pgp &>/dev/null
$ sudo chmod 644 /etc/apt/trusted.gpg.d/librealsense.pgp
```
(not /etc/apt/keyrings)

```
$ sudo apt-get install apt-transport-https
```
(no change)

```
$ echo "deb [signed-by=/etc/apt/trusted.gpg.d/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | sudo tee /etc/apt/sources.list.d/librealsense.list
```
(pointing to /etc/apt/trusted.gpg.d/ instead of /etc/apt/keyrings)

MartyG-RealSense, 2023-06-29:
Thanks very much @Comnenus for the report and your solution. 

Was your signing problem similar to the one at https://github.com/IntelRealSense/librealsense/issues/11878#issuecomment-1578375368 that occurred a few weeks ago and was resolved?

Comnenus, 2023-06-29:
Hi @MartyG-RealSense ,

I did not come across that particular post, but it does look identical.

MartyG-RealSense, 2023-06-29:
Thanks again for your report.  If we receive further reports of signing problems from other RealSense users in the next day or so then I will highlight the issue to my Intel RealSense colleagues.

MartyG-RealSense, 2023-07-08:
There were no further reports from other RealSense users of signing problems after 1 week.  

Has your problem been resolved or do you require further assistance with this case, please?  Thanks!

Comnenus, 2023-07-09:
We are all set for now, thank you!

closed 2023-07-09 by Comnenus

MartyG-RealSense, 2023-07-09:
That's great to hear, thanks very much for the update!