realsenseai/librealsense #11955: How to change camera default setting

Godofcong-1, 2023-06-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D145 | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Win11  | 
| Platform                                 | PC |
| SDK Version                            |  2.54.1                         | 
| Language                            |  python                         | 
| Segment			| yolo                         | 

### Issue Description
How to change camera hardware default setting from
```
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
```
to
```
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 60)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 60)
```

### Detailed Description

I am using YOLOv8 for continuous object detection and tracking in a camera. My current way is :

```
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 60)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 60)

......

# Streaming loop
try:
    start = time.time()
    while True:
        frames = pipeline.wait_for_frames()

        aligned_frames = align.process(frames)

        aligned_depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()

        if not aligned_depth_frame or not color_frame:
            continue    

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # get yolo
        yolo_results = yolo_model.predict(source=color_image, device="0")
```

But this wastes resources and time in transmission ( I guess ), make the actual fps only around 35 and I need higher.

So I want to try let yolo use camera directly, like this :

```
        # old get yolo
        # yolo_results = yolo_model.predict(source=color_image, device="0")
        # yolo_boxes = yolo_results[0].boxes

        # new get yolo
        yolo_results = yolo_model.track(source="0", device="0", stream=True) 
        for r in yolo_results:
            yolo_boxes = r.boxes

```

But the camera default setting is fps 30. How can I change the default setting to fps 60?

MartyG-RealSense, 2023-06-30:
Hi @Godofcong-1  The default stream profile of RealSense cameras cannot be altered.  A custom configuration has to be defined with cfg instructions like you have done.

If the actual FPS is only 35 instead of 60 then you could try disabling the RGB option **Auto-Exposure Priority**.  If auto-exposure is enabled and auto-exposure priority is disabled then the SDK will try to enforce a constant FPS instead of permitting FPS to vary.  An example of setting **auto_exposure_priority** to '0' to disable it can be found in a Python script at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360037286634-Strange-FPS-when-enable-both-color-and-depth-streams

Godofcong-1, 2023-07-03:
Thanks for help! I tried auto_exposure_priority to '0' but FPS only to 37. Maybe I should find another way. I will close this issue.

closed 2023-07-03 by Godofcong-1

MartyG-RealSense, 2023-07-03:
You are very welcome, @Godofcong-1 - thanks very much for the update.