realsenseai/librealsense #11956: Unsupported kernel version

ADgai234, 2023-06-30:
I am using ubuntu18.04 melodic (on jetson nano4G)
I ran the command ./scripts/patch-realsense-ubuntu-lts.sh
error:Unsupported kernel version 4.9.337-tegra . The Bionic patches are maintained for Bionic Beaver LTS kernels 4.1[5/8], 5.[0/3/4]
Please tell me how to solve it.Thank you very much.

MartyG-RealSense, 2023-06-30:
Hi @ADgai234  Jetson boards have their own dedicated L4T patch script.  Please try the command below.

`./scripts/patch-realsense-ubuntu-L4T.sh`

MartyG-RealSense, 2023-07-08:
Hi @ADgai234  Do you require further assistance with this case, please?  Thanks!

ADgai234, 2023-07-08:
Sorry,I was at school.Thank you very much.

MartyG-RealSense, 2023-07-08:
It's no problem at all.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-07-08 by MartyG-RealSense

tugbakara, 2023-07-21:
> 
I am trying to install SDK in ubuntu 18.04  and I am running this script but it still running does it run two,three hour or more?

MartyG-RealSense, 2023-07-21:
Hi @tugbakara  The L4T.sh script is designed for Nvidia Jetson computing boards with an Arm64 architecture.  

If you have an amd64 PC then you should be using the **./scripts/patch-realsense-ubuntu-lts.sh** script if your kernel is older than 5.13 or use **/scripts/patch-realsense-ubuntu-lts-hwe.sh** if your kernel is 5.13 or 5.15.

tugbakara, 2023-07-21:
Yeap, Actually this is what I  run I just didn't recognize file name but that's the one you corrected me. But it takes too much time to complete I guess, does it?

MartyG-RealSense, 2023-07-21:
I do not know how long it is meant to take but 2 to 3 hours to run a kernel patch sounds as though the script has frozen up.

Ideally the kernel should be patched but it is possible to use the SDK without patching.  Alternatively, you could build the SDK from packages instead of source code (as the patch is included in the packages and so you do not need to patch), or you could build from source code with the **RSUSB backend** method so that patching is not required because RSUSB bypasses the kernel.