realsenseai/librealsense #11957: hardware_reset generates "failed to set power state"  in C++ application

jpfsaunders, 2023-06-30:
|---------------------------------|------------------------------------------- |
| Camera Model   = D455
| Firmware Version  = 5.13.00.50
| Operating System & Version Linux = Ubuntu 18.04
| Kernel Version (Linux Only)  =  ?
| Platform  = nVidia Jetson Nano
| SDK Version   = ? 
| Language = C++                          | 
                         | 

### Issue Description
My C++ app operates three D455 cameras.  Normally it starts and runs correctly.  I would like to add a feature that monitors some info and then may decide to perform a hardware reset on a single camera (not all three), and then re-enable the steams and re-start the pipeline for that one camera (leaving the other two running as they originally were).  

My attempt to do this is not working.  All I did was run my code to determine if a reset is warranted (this part does work), and if it is, perform a hardware reset and then repeat the same code I used to originally (successfully) start up the camera.
When I try the hardware_reset(), the OS throws an exception "failed to set power state" and the app stops.  The reset/restart code I tried using appears below:

	// Below, variables g_list_of_cameras, serials, and ctx have been previously defined before this code block executes
	// This next block assumes that the devices of g_list_of_cameras are in the order Camera 1, 2, 3.  If not, it won't work. 
	// Note that each "for" loop should only do anything for 1 device and skip all others:
	int cn = 0;
	for (auto&& dev : g_list_of_cameras)
	{
		cn += 1;
		if (cn != camera) continue;

		cout << "resetting camera #" << camera << ", please wait ..." << endl;
		dev.hardware_reset();
		rs2::device_hub hub(ctx);
		dev = hub.wait_for_device();				
		sleep(3);
	}

	// Re-Start streaming pipe for the reset device 
	cn = 0;
	for (auto&& serial : serials)
	{
		cn += 1;
		if (cn != camera) continue;

		cout << serial << "\n\r";
		rs2::pipeline pipe(ctx);
		rs2::config cfg;
		cfg.enable_device(serial);     
		cfg.disable_all_streams();	 
		waitKey(10);

		cfg.enable_stream(RS2_STREAM_COLOR,    _FWIDTH, _FHEIGHT, RS2_FORMAT_BGR8, _CAM_FRAME_RATE);  
		cfg.enable_stream(RS2_STREAM_DEPTH,    _FWIDTH, _FHEIGHT, RS2_FORMAT_Z16,  _CAM_FRAME_RATE);  
		cfg.enable_stream(RS2_STREAM_INFRARED, _FWIDTH, _FHEIGHT, RS2_FORMAT_Y8,   _CAM_FRAME_RATE);  

		pipeline_profile selection = pipe.start(cfg);
	}

Should it be possible to use hardware_reset() on a camera more than once after initial powerup?  What might be wrong with what I am doing?

MartyG-RealSense, 2023-06-30:
Hi @jpfsaunders  There is not a C++ example of resetting a single specific camera by its serial number, though a RealSense team member provided one for Python at https://github.com/IntelRealSense/librealsense/issues/5428#issuecomment-564482167 that takes the approach of generating a list of all attached cameras with the **ctx.query_devices()** instruction and then querying a serial number in that list.

There is also a C++ reset script at https://github.com/IntelRealSense/librealsense/issues/9287#issuecomment-867826974 that cycles through all attached cameras.  This script also uses ctx.query_devices()

jpfsaunders, 2023-06-30:
Hi @MartyG-RealSense ,
Thank you for the quick response.  The Python example is exactly what I am doing in C++;  for me it only works once on powerup and then if I try it a second time after that I get an error.  I will look through the C++ reset script to see what they are doing but it looks more involved so it may take me a bit to figure out what is happening there.

MartyG-RealSense, 2023-07-01:
Thanks very much, @jpfsaunders - I look forward to your next report.  Good luck!

MartyG-RealSense, 2023-07-14:
Hi @jpfsaunders  Do you have an update about this case that you can provide, please?  Thanks!

jpfsaunders, 2023-07-14:
Hi @MartyG-RealSense ,
I have not made any progress, and this particular issue has been bumper down on my priority list.  The C++ script does not appear to be doing anything different than what I am already doing.  About the only thing I can think of is to try to reset the USB controller on my hardware (Jetson Nano) separately, either before or at the same time as performing the hardware_reset().  I am not sure how to do that, but I will experiment once I can get back to looking at this.

I would say we can close this issue.  I will open a new one later if I can't figure out a workaround.

closed 2023-07-14 by jpfsaunders

MartyG-RealSense, 2023-07-14:
Okay, thanks very much @jpfsaunders for the update.