realsenseai/librealsense #11963: D415 Showing up as D410-  ISP Update Failed

sivashankar28, 2023-07-03:
I have tried downgrading and then upgrading firmware through this post (https://github.com/IntelRealSense/librealsense/issues/3785) but I am still having issues.

MartyG-RealSense, 2023-07-04:
Hi @sivashankar28  I would advise against performing multiple downgrades and upgrades, as RealSense cameras only have an allowance of 20 downgrades before the ability to downgrade becomes locked and from that point you can only upgrade to newer firmware versions than the currently installed one.

Which firmware version is currently installed in your D415, please?  And which RealSense SDK (librealsense) version are you using with it?

sivashankar28, 2023-07-04:
D415 Firmware: 5.15.0.2 
Intel Realsense version: 2.54.1

MartyG-RealSense, 2023-07-04:
Has your D415 ever been detected as a D415 since it was purchased, or has the problem only recently occurred?

sivashankar28, 2023-07-04:
Yes it did 
It was only recently that I think my update went wrong and it’s now detected as D410

MartyG-RealSense, 2023-07-04:
Thank you.  Is your D415 a cased USB camera or the circuit board module version (below)?

[image]

sivashankar28, 2023-07-04:
It is cased

MartyG-RealSense, 2023-07-04:
If your firmware is 5.15.0.2 (which is the ideal match for SDK 2.54.1) then the camera should be working normally.  If the camera is identified as D410 then this suggests that the firmware cannot find the camera's RGB color sensor.  

I will highlight this issue to my Intel RealSense colleagues to discuss it with them.  Thanks very much for your patience!

MartyG-RealSense, 2023-07-06:
Hi @sivashankar28  When you downgraded your firmware, which version did you downgrade to?

sivashankar28, 2023-07-06:
5.9.2

MartyG-RealSense, 2023-07-06:
Ok, thank you very much.  Firmware 5.9.2 is the second oldest firmware released for the 400 Series cameras, dating back to February 2018. which is soon after the D415's launch.  A D415 should still be capable of operating with this firmware though.

I will pass your firmware downgrade details to my colleagues.  Thanks again.

MartyG-RealSense, 2023-07-07:
My colleagues tested upgrading and downgrading to firmware 5.9.2 and were not able to reproduce this problem.  Discussions with them are continuing.

MartyG-RealSense, 2023-07-19:
Hi @sivashankar28  Thanks very much for your patience.  My Intel RealSense colleagues advised that you could open up the camera and replace the **Vision Processor D4 V1** board inside it with a new one purchased by you.  It has to be the V1 version of the D4 board in order for it to work.

Otherwise, if you are unable to do this then you can send your D415 to Intel for recovery or replacement.

sivashankar28, 2023-07-19:
What could I have done to cause this issue?

MartyG-RealSense, 2023-07-19:
It may not necessarily be caused by something that you have done.  The firmware driver identifies the camera model from the components inside the camera (RGB sensor, IMU, etc) that it is able to detect.  If a fault developed in the camera hardware that prevented the RGB sensor from being detected by the firmware then it might identify the camera as a D410 (like a D415 but without RGB).

MartyG-RealSense, 2023-07-25:
Hi @sivashankar28  Do you have an update about this case that you can provide, please?  As mentioned above at https://github.com/IntelRealSense/librealsense/issues/11963#issuecomment-1641846056 you could either attempt to repair the camera yourself or you could send it to Intel for recovery or replacement.

MartyG-RealSense, 2023-07-31:
Hi @sivashankar28  Bearing in mind the advice in the comment above, do you require further assistance with this case please?  Thanks!

sivashankar28, 2023-07-31:
Apologies for the late response
I’m trying to understand, if I’ve done something to cause this and if you are able to cover the costs somehow.

MartyG-RealSense, 2023-08-01:
It does not sound as though it is a problem that you caused yourself.

I will consult with my Inel RealSense colleagues about the next step regarding sending the camera to Intel for recovery or replacement.

MartyG-RealSense, 2023-08-13:
Hi @sivashankar28  I would like to remind you to send a message to my email address in the comment above if you would like to send your camera back to Intel so that I can advise you how to proceed with that return.  Thanks!

MartyG-RealSense, 2023-09-14:
Hi @sivashankar28  Intel have repaired your D415 (see the image below).  I will provide further details by email.

[image]

MartyG-RealSense, 2023-09-21:
Hi @sivashankar28  Can you confirm whether you have received your repaired D415 please?  Thanks!

MartyG-RealSense, 2023-09-21:
That's great to hear.  Thanks so much for the confirmation.  As the D415 is received and working, I will close this case.  Thanks again!

closed 2023-09-21 by MartyG-RealSense