realsenseai/librealsense #11965: need help to solve the error below and run the code

fatteexdd, 2023-07-05:
using intel realsense lidar camera l515
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MartyG-RealSense, 2023-07-05:
Hi @fatteexdd  The error **Couldn't resolve requests** means that the camera could not provide the stream configuration that was requested.

You could remove **config** from the brackets of the start line so that the program uses the default depth and color resolution / FPS of the L515.

`self.pipeline.start()`

Could you also try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  A characteristic of the L515 model is that it works best when the micro-sized connector is in one particular orientation of the two insertion directions available.

fatteexdd, 2023-07-07:
it doesnt work

fatteexdd, 2023-07-07:
[image]
[image]
could this be the issue?

MartyG-RealSense, 2023-07-07:
The problem seems be be occurring with the **config** configuration in the early **__init__** section of the script.

What happens if the **config.resolve** line is commented out in order to remove a check for whether a particular stream profile can be provided.

[image]

fatteexdd, 2023-07-10:
my mediapipe feed can open now but it just works like a normal camera. I could not get any skeleton trackings.

MartyG-RealSense, 2023-07-10:
On the MediaPipe GitHub support forum at https://github.com/google/mediapipe/issues/3290#issuecomment-1594187876 advice was given 3 weeks ago at the time of writing this that they have no plans for MediaPipe to support RealSense.

Despite this statement, in 2022 someone published a guide for using MediaPipe and RealSense for hand detection at the link below.  It was with a RealSense D435 camera model though, but the script's 640x480 resolution for depth and color will work on L515 too.

https://medium.com/@smart-design-techology/hand-detection-in-3d-space-888433a1c1f3

closed 2023-07-24 by MartyG-RealSense