realsenseai/librealsense #11967: D435 FPS doesn

bhomaidan1990, 2023-07-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.15.00.02 | 
| Operating System & Version  | Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-07-05:
Hi @bhomaidan1990  Comparing frames to time to calculate an FPS value may not be accurate.  At https://github.com/IntelRealSense/librealsense/issues/7488#issuecomment-704124850 a RealSense team member provides technical information about how FPS is calculated in the RealSense SDK.

You could instead retrieve the camera metadata parameter **ACTUAL_FPS**

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#employing-metadata-attributes-in-demos

bhomaidan1990, 2023-07-05:
@MartyG-RealSense Thanks

closed 2023-07-05 by bhomaidan1990