realsenseai/librealsense #11970: RGB camera in RealSense Viewer not working properly

EleFontana, 2023-07-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415C | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |    Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  5.15.0-67-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                          | 
| USB Type Descriptor                           |  3.2                          | 

### Issue Description
Hi everyone.
I'm a newbie with RealSense systems, but I'm tring to use a RealSense D415 camera.

After the installation (https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md) I'm facing some problems with the RGB camera in RealSense Viewer.

When I launch the application, if I enable only RGB Camera, I can't see anything.
[image]

If I also enable Stereo Module I can see something (it still seems to me that there is something wrong).
[image]

When I disable Stereo Module again and keep moving, it shows fragmented (unfortunately, it is hard to explain).
[image]

The problem then occurs only if RGB camera is enabled by itself.

I have also tried running a calibration (https://www.youtube.com/watch?v=pcoYc4YPVMY) but I can't complete it because an error comes up: 
```
invalid value in rs2_run_on_chip calibration cpp
``` 
Has anyone had this error before and solved it?
Thank you all

MartyG-RealSense, 2023-07-05:
Hi @EleFontana  You are in the 3D **point cloud** mode of the Viewer.  At least the Depth needs to be enabled in this mode to see an image, as the pointcloud is based on depth data.  Once both Depth and RGB are enabled, the depth data of the pointcloud has RGB texture mapped onto it.

To view the RGB image on its own, you should click the **2D** option in the top corner of the Viewer's window.

EleFontana, 2023-07-05:
Oh, my bad.

In fact, clicking on the 2d option everything shows correctly. 

Thank you!

closed 2023-07-05 by EleFontana