realsenseai/librealsense #11973: [Intel RealSense D435 Python API, Ubuntu 22.04] frames = pipeline.wait_for_frames() RuntimeError: Frame didn't arrive within 5000

monajalal, 2023-07-06:
Hi @MartyG-RealSense thanks a lot for all your supports in the issues.
I have no problem running this code with my laptop at the office but we use a USB extender over there (which I don't have with me now). When I directly connect the USB cable to either of USB ports on my machine, I get the following error for the code below:
```
(calib) mona@ard-gpu-01:~/intel_realsense_captures$ cat capture.py 
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

color_path = 'V00P00A00C00_rgb.avi'
depth_path = 'V00P00A00C00_depth.avi'
colorwriter = cv2.VideoWriter(color_path, cv2.VideoWriter_fourcc(*'XVID'), 30, (640,480), 1)
depthwriter = cv2.VideoWriter(depth_path, cv2.VideoWriter_fourcc(*'XVID'), 30, (640,480), 1)

pipeline.start(config)
profile = pipeline.get_active_profile()
color_intrinsics = profile.get_stream(rs.stream.color).as_video_stream_profile().get_intrinsics()

try:
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue
        
        #convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        
        colorwriter.write(color_image)
        depthwriter.write(depth_colormap)
        
        cv2.imshow('Stream', depth_colormap)
        
        if cv2.waitKey(1) == ord("q"):
            break
finally:
    colorwriter.release()
    depthwriter.release()

        # Print the intrinsic parameters
    print("Color Intrinsics:")
    print("Width:", color_intrinsics.width)
    print("Height:", color_intrinsics.height)
    print("Principal Point (x, y):", color_intrinsics.ppx, color_intrinsics.ppy)
    print("Focal Length (x, y):", color_intrinsics.fx, color_intrinsics.fy)

    pipeline.stop()
```
as you see it shows the first frame and then throws the following error:
```
(calib) mona@ard-gpu-01:~/intel_realsense_captures$ python capture.py 
Color Intrinsics:
Width: 640
Height: 480
Principal Point (x, y): 321.5348815917969 235.87950134277344
Focal Length (x, y): 616.056884765625 616.2793579101562
Traceback (most recent call last):
  File "/home/mona/intel_realsense_captures/capture.py", line 22, in <module>
    frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
```
[image]

[... 119 lines cut ...]

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-07-07:
You are very welcome, @monajalal   :)

How long is your extender cable, please?  If it is longer than 2 meters then it is important to use a high quality cable that is designed for data transfer because of the high volume of data that RealSense cameras can transmit.

If the cable is longer than 5 meters then I would recommend one with **active repeater** components in it to boost the signal over distance.  The company **Newnex** sell high quality USB cables that have been validated for use with RealSense.

https://www.newnex.com/realsense-3d-camera-connectivity.php

monajalal, 2023-07-07:
Thanks for your response. I am using the same exact camera and cable that I used in the office. I just forgot to bring the USB extender we connect for our mouse and cameras.

I am curious if the USB ports on Thinkpad laptop may cause some problem?
The cable is as shown in the photos and is a bit over 1m. 
I will check again when I am back in the office on Monday and update you and share more details about the USB extender I've been using.

[image]
[image]

MartyG-RealSense, 2023-07-07:
Thank you very much for the details and images.

I use a ThinkPad like yours with RealSense cameras extensively myself.  It is compatible with all RealSense models, though I find that some specific camera units may disconnect and reconnect.

monajalal, 2023-07-11:
I'm gonna close this issue since when I use this usb connector, I don't have the problem
[image]

closed 2023-07-11 by monajalal

MartyG-RealSense, 2023-07-11:
It's great to hear that you found a solution, @monajalal - thanks very much for the update!

animesh-singhal, 2024-01-19:
@monajalal: Thanks! A USB Hub also fixed my issue. 
@MartyG-RealSense: I use AMD Ryzen 9, so the processor isn't the problem.

MartyG-RealSense, 2024-01-19:
Hi @animesh-singhal  Thanks very much for the confirmation that a USB hub worked for you too!